greatscottgadgets/hackrf #784: Restore OFW from Mayhem

cyb3rr0nn13, 2020-09-04:
does anyone know how to get back from the portapack mayhem firmware to the original from mike ossmann?

**The issue:**
if i connect the HackRF to my Raspberry Pi with the Mayhem FW installed, but without portapack attached, it will not recognize....

how do i restore the ofw? thanks

miek, 2020-09-04:
You should be able to follow the DFU boot instructions here: https://github.com/mossmann/hackrf/wiki/Updating-Firmware#only-if-necessary-dfu-boot to load the firmware image temporarily, then follow the normal flashing instructions from further up the page (using `hackrf_spiflash`) to restore the original firmware.

cyb3rr0nn13, 2020-09-04:
ok, long story short:

-boot hackrf
-press DFU button
-press RESET while holding DFU
-now only one red LED is active
-execute `dfu-util --device 1fc9: 000c --alt 0 --download hackrf_one_usb.dfu` in a terminal

and thats it?

miek, 2020-09-04:
Those are the first steps, and they should get the device showing up and responding to `hackrf_info` etc. but it's only temporary at that point. If you reset it after that, it'll go back to the way it was.
So, once you've run those steps, then you want to use `hackrf_spiflash` (https://github.com/mossmann/hackrf/wiki/Updating-Firmware#step-1-updating-the-spi-flash-firmware) to fully flash it so it'll persist after a reset.

cyb3rr0nn13, 2020-09-04:
thanks bro.....ill try in a few mins.....have to do some other shit first!

cyb3rr0nn13, 2020-09-04:
UPDATE:
ive tried, but somehow it doesnt recognize the device....
even `dmesg` doesnt show anything.....
and `dfu-util` shows the message ive screenshotted.....

[image]

[image]

whats the reason? maybe wrong device-string?

Radiomix2000, 2020-09-06:
of course first of all you need to set your portapack to HACKRF MODE. Otherwise you can't update

cyb3rr0nn13, 2020-09-06:
...did so....

miek, 2020-09-09:
If it's not coming up in dmesg when started in DFU mode, and you've tried different USB cables/ports/etc., then I'd suspect there's a problem with the hardware.

cyb3rr0nn13, 2020-09-10:
but how can i get rid of this shit?
i mean, now its running quite well, but ONLY WITH PORTAPACK...and only on MAYHEM 1.2.....
i dont see a possibility to update the device in a few days without any working usb connection....
 
i dont know what i did wrong, just unplugged the device from my pi and after i reconnected, i realized theres no fucking hackrf shown......please, tell me how to restore.....i thought about a buspirate or something like that to get connected.....what you think?

`not-root@me-not:~/software/hackrf/firmware/hackrf_usb $ dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb_dfu.bin
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available
`

miek, 2020-09-10:
When you boot the HackRF in DFU mode, that bypasses all firmware. If it doesn't show up on USB in that mode, then there must be some damage to the hardware (either cables, the USB connector, ESD protection, or the microcontroller itself). It may be repairable if you want to get into that, but it might be best to look at RMA options.

Which version of the HackRF / PortaPack do you have? Can you post a picture or link to the product page?

cyb3rr0nn13, 2020-09-11:
thats really bad news......300€ "blowin in the wind"
but how the fuck could this happen? did nothing wrong so far.....at least i think so

Radiomix2000, 2020-09-11:
> 
> 
> ...did so....no device registered...

you mean hackrf_info doesn't produce any output regarding hackrf serial number etc..???

cyb3rr0nn13, 2020-09-11:
oh.....no.....it wont recognize anything.....
not on windows and not on linux.....

miek, 2020-09-11:
> thats really bad news......300€ "blowin in the wind"
> [...]
> @miek sure....
> the hackrf i bought [here](https://www.wimo.com/de/hackrf-one)

Don't worry, get in touch with them and let them know about the problem & this thread, and they'll replace it for you.

> but how the fuck could this happen? did nothing wrong so far.....at least i think so

For that, we're still investigating. It would actually help a lot if you could post some pictures of the PortaPack H2 that you have. Particularly of the ICs around the battery connector/audio jack area.

cyb3rr0nn13, 2020-09-11:
OK, i will try to get in contact with them on Monday....
and here are the pictures of my PortaPack H2

[image]

[image]

miek, 2020-09-11:
Thanks!

Are there any markings visible on this IC?
[image]

miek, 2020-09-11:
A couple more questions if you don't mind :)
Were you using the PortaPack H2 with the internal battery?

and do you have a multimeter? If so, could you measure the output voltage from the PortaPack H2 (with the battery connected, but not with it attached to the HackRF):
[image]
(that's the bottom left pin on that header, and you can get ground on the mounting hole)

cyb3rr0nn13, 2020-09-12:
Nope, no battery and no multimeter.....
have to buy a battery but dunno where.....

vasja2000, 2020-09-15:
> Thanks!
> 
> Are there any markings visible on this IC?
> [image]

ETA6102 6362C

cyb3rr0nn13, 2020-10-11:
Good news folks ... just for fun I tried to see if the HackRF mode works ... and lo and behold, suddenly everything works again ... please don't ask me what it should have been now, but EVERYTHING WORKS AGAIN !!! ONLY DO NOTHING FROM A FEW DAYS .... MADE MY DAY!

UPGRADE IM COOOMINNG!

closed 2020-10-11 by cyb3rr0nn13