greatscottgadgets/hackrf #786: make: *** [all] Error 2

mieluw, 2020-09-07:
I am using Ubuntu 18.04 LTS 64 bit
cmake 3.7.2
Python 3.6
Pip 3

when i try to makefrom this source https://github.com/mossmann/hackrf/tree/master/firmware it shows 
[image]

CMakeFiles/hackrf_usb_objects.dir/build.make:326: recipe for target 'CMakeFiles/hackrf_usb_objects.dir/usb_api_board_info.c.obj' failed

@postmodern @posixninja

miek, 2020-09-09:
Could you post the full output from the cmake command?

straithe, 2021-01-26:
Are you still having trouble? We will close this issue if we don't get a reply within the next month.

closed 2021-02-25 by straithe