greatscottgadgets/hackrf #787: Question: how to log signal above a certain threshold

larkolab, 2020-09-08:
Hello,
I'm not sure if we can ask question here, so sorry if it is the not the proper place.

I would like to know if there is a command tool which would allow to set a signal strength threshold and capture all the moments where the signal is above that threshold (with timestamp). All this in a given frequency range.

The use case is that I would like to know when a device transmits over a quite long period (several hours).

I've looked at rx_tools/rx_power, we we cannot set the threshold trigger.

Thanks in advance.
Best regards,
Michael

miek, 2020-09-09:
I don't know of any off-the-self tools to do it.
One option would be to write a GNURadio flowgraph for it. It should be possible using the 'Burst Tagger' and 'Tagged file sink' blocks.

larkolab, 2020-09-10:
Hi @miek ,
Thank you for your answer, I'll have a look to gnuradio and see if I can manage to set it up. :)
Best regards,
Michael

larkolab, 2020-09-10:
@miek Would you have a flowgraph example, I really don't know how to configure the Burst Tagger :/ And I'm not a signal processing expert, so lotss of unknown. :/
I have GRC running, and I'm able to see the signal with a QT GUI Sink. But then, from there, detecting the signal and logging, I have no idea how to configure this.

miek, 2020-09-10:
Sure, here's an example:
[image]
That generates a signal with a varying amplitude and writes out a file for each period that it's above the threshold.

[taggedfilesink.zip](https://github.com/mossmann/hackrf/files/5204487/taggedfilesink.zip)

larkolab, 2020-09-11:
@miek thank you very much !

closed 2020-09-11 by larkolab