greatscottgadgets/hackrf #788: Can a single HackRF perform RX/TX in GNURadio based on button press (PTT)

metmayhem, 2020-09-10:
I recently bought a hackrf and have been going through the lessons. I have a question and was told to ask it here. Is it possible in gnuradio to have a single HackRF flip back and forth between RX and TX using a button to establish PTT? I tried without any success so far, but thought maybe someone might have more information. Or maybe there is different software I should use?

GravesJake, 2020-11-11:
Hi, sorry this fell through the cracks. Are you still having trouble accomplishing this? I personally haven't tried this but it should definitely be feasible.

straithe, 2021-02-05:
There hasn't been a response on this issue for a while. Do you still have this question? If you don't respond within a month I will close this issue.

metmayhem, 2021-02-05:
Yes, I'm still having trouble with this. I kind of set this one aside and have been working on other projects, but would really like to get this working, if possible. Any ideas or things to try would be appreciated.

miek, 2021-02-06:
There was a talk recently at the GNURadio Amateur Radio Meetup where Gavin Jacobs went through his solution to this: https://www.youtube.com/watch?v=rTp6vQk5MqE

closed 2021-03-18 by straithe