greatscottgadgets/hackrf #789: Problem on GnuRadio only with TXCO

do1kbl, 2020-09-12:
### Steps to reproduce
1. Make on GnuRadio a project with Osmocom Source block to record the HF as File
2. If a TXCO connected (0x01) then the Signal its totally overridden. Without TXCO (0x51) its all OK.
3. Make on GnuRadio a project with Osmocom Sink to transmit the recorded file

### Expected behaviour
I would have expected everything to work exactly as it would without the TXCO

### Actual behaviour   Tatsächliches Verhalten
The output its totally overridden.
When I record without TXCO and replay, everything its ok.
when I replay the recorded file with TXCO, i head nothing only overridden.
when I record with TXCO and playback its also Ovenridden.

On my scope I checked the TXCO it has 10MHZ Square from 0 to 3,2V

When I disconnect the TXCO and connect 10MHZ square on ClockIn from Generator with 3,3V its the same problem.
I try it on Two hackRF. With Portapack with TXCO its the same probleme.

But on SDR# I can hear Everything without problem, no different between TXCO is connected or not.

### Version information 
**Operating system**:
Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.5 LTS
Release:	18.04
Codename:	bionic

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000046d067dc286c1847
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006f4f4e

**Gnu Radio Version**
GNU Radio Companion 3.7.11

GravesJake, 2020-11-11:
Hi, are you still having TXCO problems?

do1kbl, 2020-11-13:
yes its not working

straithe, 2021-05-10:
Sorry about the delay in responding, are you still experiencing these issues?

do1kbl, 2021-05-10:
Yes I have still the same Problem
You have any Ideas ?

miek, 2021-05-10:
I had a look at your GRC file and I notice that the sample rate is set to 32ksps. That'll certainly be causing problems as the minimum supported rate for the HackRF One is 2Msps (and it's recommended to operate at 8Msps or above for best image rejection).

straithe, 2021-05-31:
@miek is correct, HackRF doesn't support anything under 2Msps. Please set your sample rate above 2Msps and let us know if you are still having those issues.

do1kbl, 2021-06-02:
I will Test it tomorrow.
Where I can upload my recorded files? Ant Idea?

do1kbl, 2021-06-09:
Today I test ist again with 4msps
Same problem

I record a FM signal it was overridden 
I record a DMR signal it’s was broken on playback. The device can only decode some
Voice snippets.

when I disconnect the tcxo all it’s fine.
It’s possible with a command to deactivate the tcxo temporally? Connect end disconnect it’s not perfect solution but
Maybe I can ignore it when I used gnuRadio

straithe, 2021-06-14:
Have you tried different TCXOs? It sounds like the one you have may be faulty.

do1kbl, 2021-06-14:
Yes I try portapack with txco and external txco on two different hackRF 

Same problem

straithe, 2021-06-14:
Have you tried different styles, brands, and types of external TCXOs?

do1kbl, 2021-06-14:
Yes.
Without its all ok, same problem I have if I connect a second hackRF with clock out to clock in.

I have no idea what is the problem, it’s possible to deactivate the txco in software ? That he used the internal. It’s better as open the cabinet disconnect the txco and closed

mossmann, 2021-06-15:
> I have no idea what is the problem, it’s possible to deactivate the txco in software ?

The firmware does not provide a way to disable clock input from software.

Could you please add a QT GUI Frequency Sink to your flowgraph and show us screenshots with and without clock input?

do1kbl, 2021-06-15:
I will try 👍🏻

do1kbl, 2021-06-17:
So here you have 3 Zip files
Picture are screenshots
Record is the recording files
projects are my GnuRadio Projects.

I make with the Same HackRF. With TXCO then I disconnect and close the Cabinet and make the Same Test without TXCO
Maybe any Ideas.

In the Project you need to change the Wavfile name and frequency maybe.

[pictures.zip](https://github.com/mossmann/hackrf/files/6668736/pictures.zip)
[Record.zip](https://github.com/mossmann/hackrf/files/6668737/Record.zip)
[projects.zip](https://github.com/mossmann/hackrf/files/6668738/projects.zip)

mossmann, 2021-06-17:
Thank you for the screenshots! It looks like you are not using offset tuning, so the signal of interest is directly on top of the DC spike. This is worse when the TCXO is in use because of the improved frequency accuracy. Can you please try changing the tuning frequency in your flowgraphs from 433.512 MHz to 434 MHz?

do1kbl, 2021-06-17:
I will check this tomorrow

mossmann, 2021-06-22:
Yay!  I'm so glad that worked!  Offset tuning is an important technique in SDR; it is suggested here: 
https://github.com/mossmann/hackrf/wiki/FAQ#how-do-i-deal-with-the-dc-offset
as a solution for: https://github.com/mossmann/hackrf/wiki/FAQ#how-do-i-deal-with-the-dc-offset

For a more general education in SDR: https://greatscottgadgets.com/sdr/

closed 2021-06-22 by mossmann