greatscottgadgets/hackrf #790: hackrf_open() failed: Entity not found(-1000)

efakhri, 2020-09-15:
### Steps to reproduce
1. Plug HackRF in
2. Launch `hackrf_info`
### Expected behaviour
```
Found HackRF
Index 0
Serial number: -----------------
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: ----------------
```

### Actual behaviour
```
Found HackRF
Index 0
hackrf_open() failed: Entity not found (-1000)
```

### Version information
**Operating system**:
Ubuntu 18.04
**hackrf_info output:**
```
Found HackRF
Index 0
hackrf_open() failed: Entity not found (-1000)
```

GravesJake, 2020-11-11:
Are you still having this issue?

efakhri, 2020-11-12:
Yes. I have tried different usb cables and different ports but still same. lsusb shows 
`Bus 003 Device 006: ID 1d50:6089 OpenMoko, Inc.`

GravesJake, 2020-11-12:
Have you tried running `hackrf_info` as root?

5ch4um1, 2021-01-05:
I think i have the same issue.

`:~$ hackrf_info 
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
hackrf_open() failed: Input/Output Error (-1000)`

 but if i run it again, it works: 
`~$ hackrf_info 
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2433951b
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0041475c`

same for gnuradio-companion with the osmocom source, the first time i click on run, i always get:

`RuntimeError: Failed to open HackRF device (-1000) No such device (it may have been disconnected)`
and if i click again, it works:
`Using HackRF One with firmware 2017.02.1`

udev rules are set, OS is Ubuntu 18.04.
Unfortunately this "click again" workaround doesn't seem to work with gqrx, would be nice to find out what causes this?

straithe, 2021-03-24:
Have you tried running the command quickly twice in a row?

straithe, 2021-03-24:
We've recently done a release which includes some enhances and fixes to hackrf_sweep. Can you please try updating your HackRF firmware to the latest release to see if that fixes your issue?

5ch4um1, 2021-03-24:
> We've recently done a release which includes some enhances and fixes to hackrf_sweep. Can you please try updating your HackRF firmware to the latest release to see if that fixes your issue?

For me this did indeed fix the issue, thank you very much! :)
tested with gqrx, cubicsdr, sigdigger and hackrf_info, all work as expected now.
Firmware Version: 2021.03.1 (API:1.04)

straithe, 2021-03-26:
Great! Thank you for responding to let me know that worked. Since we've resolved the problem you were encountering I'm going to close this issue.

closed 2021-03-26 by straithe

MAVProxyUser, 2022-03-24:
Is there a solution for this? I've got an old HackRF from 2017 that I used to have PortaPack Havoc on. Now I can't flash it via DFU, or anything really

```
$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu 
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        21992 bytes
Download done.
dfu-util: unable to read DFU status after completion
```
and
```
hackrf_info version: git-d3e4e9b
libhackrf version: git-d3e4e9b (0.7)
Found HackRF
Index: 0
Serial number: @@
hackrf_open() failed: Entity not found (-1000)
```

martinling, 2022-03-24:
Hi @MAVProxyUser, that looks to me like DFU is working OK. The `unable to read DFU status after completion` message isn't a problem I think.

However, it says the firmware you're uploading is `21992 bytes`, which if it's correct must be a very old build - current versions are around 36KB.

I see you've got the latest host tools and libhackrf from git, so I suggest building and flashing the latest firmware from there too.

If you're still having trouble after that, please open a new issue.