greatscottgadgets/hackrf #792: Installation problem，how can i make the cmake …/ ‐DINSTALL_UDEV_RULES=O run successful

Tahiti-luo, 2020-09-22:
### Steps to reproduce
1.~/SDR$ cd ~/SDR/hackrf/host
~/SDR/hackrf/host$ mkdir build && cd build
~/SDR/hackrf/host/build$ cmake …/ ‐DINSTALL_UDEV_RULES=ON
CMake Error: The source directory "/home/luo/SDR/hackrf/host/build/‐DINSTALL_UDEV_RULES=ON" does not exist.
Specify --help for usage, or press the help button on the CMake GUI.

2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

GravesJake, 2020-11-11:
Are you still having trouble with this?

straithe, 2021-01-28:
Are you still experiencing this issue? We will close this issue if we don't get a reply within the next month.

closed 2021-02-28 by straithe