greatscottgadgets/hackrf #793: Device argument not shown in osmocom source

aroo-del, 2020-09-23:
Hi, 
I am using Hackrf with GNU radio version 3.8.1. I have installed it successfully.
I have to program it to show device arguments option in osmocom source and osmocom sink but it is not shown. I didn't installed osmocom from any respiratory. it was installed along with my GNU radio. 
Can someone tell how to install osmocom such that it shows device argument option too?

GravesJake, 2020-11-11:
Are you till having trouble with this? What are you using to interface your device?

aroo-del, 2020-11-12:
No its fine now. I have done it. Sent from my Huawei Mobile-------- Original Message --------Subject: Re: [mossmann/hackrf] Device argument not shown in osmocom source  (#793)From: Jacob Graves To: mossmann/hackrf CC: aroo-del ,Author 
Are you till having trouble with this? What are you using to interface your device?

[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/mossmann/hackrf/issues/793#issuecomment-725684846",
"url": "https://github.com/mossmann/hackrf/issues/793#issuecomment-725684846",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

closed 2020-11-13 by GravesJake