greatscottgadgets/hackrf #794: WX vs QT GUI

PRoots, 2020-09-28:
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc24963a63
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004b4f4d

Using GNU Radio Companion 3.8.1.0 (Python 3.8.2)
Running on Kubuntu 20.04 (also tried on Ubuntu and Xubuntu)

I have been struggling to follow the SDR tutorials on the greatscottgadges site as the lessons and downloadable example files rely on WX GUI blocks but there are only QT GUI blocks (WX GUI is not even listed on the GNU Radio wiki). I have been able to guess some of the equivalents but not others.
In an effort to get something to work I have tried CubicSDR and gqrx but at best I can only just make out an FM station above the hiss with an ant500 and RF amp turned on. 
Are there any lesson downloads available using QT GUI or lists of equivalents I could use? 
Any suggestions as to why I have no luck with CubicSDR or gqrx?

Many thanks

GravesJake, 2020-11-11:
Are you still having trouble getting through the tutorial using QT?

PRoots, 2020-11-11:
Yes. I was getting nowhere and have rather given up, which is disappointing.

GravesJake, 2020-11-12:
Is there a particular block that you can't get to work?

PRoots, 2020-11-14:
Well I did not get past lesson 1 as the flow diagram I produced did not seem to produce a working radio - I used QT GUI Range for the sliders which were an easy match, I think.
The QT GUI frequency sink (for WX GUI FFT Sink) I found a bit tricky as the options the two had did not really match up (to me anyway)
I got a graph of the frequencies so I assumed I was on the right track but where I fell down was when the radio would not tune into any stations at all, I could not just download the example to look at and see what I got wrong. I have been looking, on and off, for QT examples on the web but most of what I have found refers to the WX GUI so not had much luck.

F4HFQ, 2020-11-14:
Hi,

Tell me if my flowgraph Can Hello toi.

Franck

PRoots, 2020-11-14:
Franck - that is brilliant. Not fantastic reception but I can tune into some stations which is more than I have been able to do before. So now I have a working model to be able to figure out where I went wrong in my flow diagram.
Many thanks for that

straithe, 2021-02-05:
@PRoots, do you still need assistance or did Franck solve your issue?

PRoots, 2021-02-13:
@straithe Thanks for following up, I think Franck got me started although I have not had time recently to try more. I think I probably need to look at a different antenna to the supplied one as I am getting very poor reception so not too easy to get things working well. On the right track now though so thanks

straithe, 2021-02-14:
Thanks for letting me know!

closed 2021-02-14 by straithe