greatscottgadgets/hackrf #795: Pipe error (-1000) on mac

theresiasnow, 2020-10-03:
### Steps to reproduce
1. latest mac os with latest hackrf mayhem and latest hackrf tools from master.

2. try to run hackrf_sweep

### Expected behaviour

a frequency scan

### Actual behaviour

Pipe error (-1000) on mac

### Version information
**Operating system**:

mac os 10.15.7 (19H2)

**hackrf_info output:**
```
 $> hackrf_info
hackrf_info version: git-6e5cbda
libhackrf version: git-6e5cbda (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000026b468dc387d6f8f
Board ID Number: 2 (HackRF One)
Firmware Version: local-883a622 (API:1.04)
Part ID Number: 0xa000cb3c 0x00464f5c
```

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
 $>  hackrf_sweep
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
hackrf_start_rx_sweep() failed: Pipe error (-1000)
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .

```

GravesJake, 2020-11-11:
Are you still having trouble with this? Have you tried running sweep with elevated permissions?

theresiasnow, 2020-11-12:
yes same error using sudo

GravesJake, 2020-11-13:
Are you able to use other commands such as `hackrf_transfer`?

theresiasnow, 2020-11-13:
yes they work

På 13 november 2020 kl. 01:18:33, Jacob Graves (notifications@github.com)
skrev:

Are you able to use other commands such as hackrf_transfer?

—
<https://github.com/mossmann/hackrf/issues/795#issuecomment-726422282>, or
unsubscribe

straithe, 2021-03-24:
We've recently made a release which includes some enhances and fixes to hackrf_sweep. Can you please try updating to the latest release to see if that fixes your issue?

straithe, 2021-04-23:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-04-23 by straithe

klaute, 2024-03-02:
Hi,

Im facing an issue with my rad1o badge (with the latest rev5 hackrf rad1o firmware) running using gnuradio companion with Osmocom support and gqrx.

hackrf_sweep works fine and also hackrf_transfer.

Except if I try to use the hackrf_transfer -p1 option.

This causes the error:
Failed to apply antenna bias voltage state: 0 (-1000) Pipe error

Without the -p1 option, I can see that the rad1o is sending.

Tested with hackrf_transfer -f<frequency> -c100

I received the sent signal on a second radio…

Sending and receiving using any hackrf cmd line tools works, no matter if it is Linux or macOS.
Also the tool version does not matter.

Now the issue I’m facing:
Only gnuradio and also GQRX complain (in the same way) about the bias voltage like -p1 would be enabled.

I tried to disable the bias by adding „bias_tx=0/1/true/false“ to the device string. I did the same for the „bias“ option. Also tried it in the osmocom sink options in GRC and also in the connection settings of GQRX.

In short I got no success. Also using sudo does not help.

How could I fix that? Any suggestions?

Thank you!

Mac gqrx:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.5.0
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf spyserver rfspace airspy airspyhf soapy redpitaya 
[INFO] [UHD] Mac OS; Clang version 15.0.0 (clang-1500.1.0.2.5); Boost_107100; UHD_3.15.0.0-MacPorts-Release
PortAudio V19.7.0-devel, revision unknown (version [1246976](tel:1246976))

Linux gqrx:
gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.10.5.1
built-in source types: file fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy airspyhf soapy redpitaya freesrp xtrx
[INFO] [UHD] linux; GNU C++ version 12.2.0; Boost_107400; UHD_4.3.0.0+ds1-5

Mac GRC:
3.8.5.0 (Python 3.9.18)

Linux GRC:
3.10.5.1 (Python 3.11.2)

Mac hackrf:
hackrf_info
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: [000000000](tel:000000000)………
Board ID Number: 3 (rad1o)
Firmware Version: git-b6acd66c (API:1.06)
Part ID Number: 0xa0000a30 0x0059434e
Operacake found, address: 0xff

Linux hackrf:
hackrf_info 
hackrf_info version: 2022.09.1
libhackrf version: 2022.09.1 (0.7)

martinling, 2024-03-02:
@klaute please open a new issue.