greatscottgadgets/hackrf #799: No Audio in FM Receiver hackrf

hussainak, 2020-10-23:
### Expected behaviour
FM GRC plays audio when tuned in

### Actual behaviour
I have 2 brand new HackRF SDRs from great scott gadgets. I followed lesson 1 and managed to create GNURadio blocks using QT GUI. I am able to execute the flowgraph and I can see some FM stations but when I tune into them I can only listen to a hiss and no audio. Is this a fault with the hardware? Please find the grc and image attached.

### Version information
**Operating system**: OSX Sierra

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000457863c8257d4f1f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00504744
[
[fm.grc.zip](https://github.com/mossmann/hackrf/files/5428977/fm.grc.zip)
[image]

](url)
If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
<<< Welcome to GNU Radio Companion 3.8.2.0 >>>

Block paths:
	/Users/ali/.grc_gnuradio
	/opt/local/share/gnuradio/grc/blocks

Loading: "/Users/ali/Desktop/Development/GnuRadio/fm.grc"
>>> Done

Generating: '/Users/ali/Desktop/Development/GnuRadio/top_block.py'

Executing: /opt/local/Library/Frameworks/Python.framework/Versions/3.7/bin/python3.7 -u /Users/ali/Desktop/Development/GnuRadio/top_block.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.2.0
built-in source types: file rtl rtl_tcp uhd sdrplay hackrf bladerf spyserver rfspace airspy airspyhf soapy redpitaya 
Using HackRF One with firmware 2018.01.1
gr::log :INFO: audio source - Audio sink arch: osx
gr::log :INFO: audio_osx_sink0 - 

Using output audio device 'Built-in Output'.
  ... which is the current default output audio device.
  Changing the default output audio device in the System Preferences will 
  result in changing it here, too (with an internal reconfiguration).

aUaUaUaUaUaUaUaUaUaU
>>> Done

hussainak, 2020-10-28:
Hi @mossmann 

I have looked into this further with SDR#. Looks like the hardware has a very weak signal and needs an amplifier. If I turn up the LNA gain on SDR# I am then able to hear the radio. Is there something you can suggest in your lesson 1 that can be manipulated to have the appropriate gain on osmocom source in gnuradio. I played around with IF gain and BB gain but was still unable to hear the FM radio band. My devices were part of the Kickstarter campaign but I only had a chance to work on them now. I know it’s been years but the spark is still alive!

0xGSMonkey, 2020-10-28:
You can change the IF or BaseBand signal parameters in proportion to
increase the gain. There is a book on RF That I can’t remember what it is
called, but once I do I will let you know. Mess around with the IF, RF and
baseband gain values to produce audio. Weak signals are hard to hear in an
electrically noisy environment!

hussainak, 2020-10-28:
Thanks @0xGSMonkey for responding,

I will give it a shot too and if I come across anything I will post it here. Appreciate your response! Thanks.

mossmann, 2021-01-26:
It looks like there is an error in the flowgraph. `freq_slider - channel_freq` in the Signal Source should probably be `channel_freq - freq_slider`. In the default configuration of this flowgraph, the FFT display is centered at 101.9 MHz, but the demodulation is centered at 98.9 MHz (1.9 MHz below the tuning frequency of 100 MHz instead of 1.9 MHz above).

closed 2021-02-23 by straithe