greatscottgadgets/hackrf #803: HackRF_one antenna problem (low freq)

2020270127, 2020-10-31:
Version information
**Operating system**:
(vmware_)ubuntu 18.0.4
**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc240e8063
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006e474f

//////////////////////////////////////////////////////////////////////////////////
hi, 
im using hackrf_one with gnu_radio program to exploit doorlock.
my doorlock uses 10~15MHz freq and i dont now exact samp_rate.

my problem is that my antenna can only catch min 70MHz to  max 2GHz.
so i searched for antenna that can catch low freq, but it's for chipwhisperer, 
although it's sma-male, it's(my new antenna) size does not match with hackrf.

1. is there a converter or someting that can use chipwhisperer's antenna on hackrf_one?
2. if not, is there a antenna that can catch low freq(i mean at least 10MHz) support on hackrf_one?

sorry for poor english, thank u

DB-Billy, 2020-12-06:
Hi 2020270127
You can use miniwhip antenna for lower frequencies (0-50mhz)

mossmann, 2021-01-26:
If the target device communicates over the near field, the Chipwhisperer H-probe should be a good solution. If the target does not use the near field, a miniwhip as @DB-Billy suggests would be appropriate for receiving signals from a distance.

closed 2021-01-26 by mossmann