greatscottgadgets/hackrf #806: Usb does not detect in any mode

sysardex, 2020-11-16:
### Steps to reproduce
1. When connecting the device to the usb port, there is no reaction on the computer. The device itself receives power and everything except usb works.
2. Trying dfu mode. Also no reaction. The device itself responds with LEDs to dfu mode, but the computer also does not see any usb device.
3. Except for usb, the device is completely working fine. It picks up the radio signal and sends it when working with portapack. More than 10 different usb wires and computers have been re-tried. I replaced LXES1TBCC2-004 just in case, but it didn't help.

### Expected behaviour
This problem appeared after I connected the hackrf to a raspberry pi power supply. Perhaps the power supply gave more voltage than 5v or its connector was defective.

### Actual behaviour

### Version information
**Operating system**:
Linux, Windows, Macos, Raspbian
**hackrf_info output:**
Nothing
### Output

Please help me understand whether my problem will be solved by replacing the LPC4320FBD144 chip or it will not help. What points on the board can I check? Thanks.

kevinlola, 2020-12-04:
Bonjour 
Mon pc Sony windows 10 ne détecte pas le hackrf one mes après avoir changer de port usb vous pouvez m'aider mes j'ai esseye sur rf analyzer qui fonctionne très bien sûr Android

kevinlola, 2020-12-04:
Bonjour
Mon pc Sony windows 10 ne détecte pas le hackrf one même après avoir changer de port usb vous pouvez m'aider merci?

 mes j'ai esseye sur rf analyzer qui fonctionne très bien sûr Android

MoJo2600, 2020-12-12:
I have the exact same issue. I can rule out the LXES1 ESD protection chip and the USB data lines are not shorted to ground or VCC. The LEDs are working as expected. I have the same question. Could this be a software problem (mine stopped working after a firmware upgrade) or a faulty LPC4320 chip.

miek, 2021-01-04:
> I have the exact same issue. I can rule out the LXES1 ESD protection chip and the USB data lines are not shorted to ground or VCC. The LEDs are working as expected. I have the same question. Could this be a software problem (mine stopped working after a firmware upgrade) or a faulty LPC4320 chip.

Does any device show up if you try booting in DFU mode (by holding the DFU button while plugging in the HackRF)?

MoJo2600, 2021-01-05:
There was absolutely no reaction on the USB port - with DFU or without. Yesterday I replaced the NXP 4320 chip on the board and it is working again! The USB connector on the board was damaged. I believe what happend is that the broken USB port destroyed the USB port inside the NXP Microcontroller.

closed 2021-02-06 by miek