greatscottgadgets/hackrf #807: no voltage out of hackrf

markl17, 2020-11-18:
### Steps to reproduce
1.turn on go into 
2.go to hackrf mode
3.type python3 followed by this code
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

#
# SPDX-License-Identifier: GPL-3.0
#
# GNU Radio Python Flow Graph
# Title: Not titled yet
# GNU Radio version: 3.8.0.0

from distutils.version import StrictVersion

if __name__ == '__main__':
    import ctypes
    import sys
    if sys.platform.startswith('linux'):
        try:
            x11 = ctypes.cdll.LoadLibrary('libX11.so')
            x11.XInitThreads()
        except:
            print("Warning: failed to XInitThreads()")

from gnuradio import analog
from gnuradio import gr
from gnuradio.filter import firdes
import sys
import signal
from PyQt5 import Qt
from argparse import ArgumentParser
from gnuradio.eng_arg import eng_float, intx
from gnuradio import eng_notation
from gnuradio.qtgui import Range, RangeWidget
import os
import osmosdr
import time
from gnuradio import qtgui

class test(gr.top_block, Qt.QWidget):

    def __init__(self):
        gr.top_block.__init__(self, "Not titled yet")
        Qt.QWidget.__init__(self)
        self.setWindowTitle("Not titled yet")
        qtgui.util.check_set_qss()
        try:
            self.setWindowIcon(Qt.QIcon.fromTheme('gnuradio-grc'))
        except:
            pass
        self.top_scroll_layout = Qt.QVBoxLayout()
        self.setLayout(self.top_scroll_layout)
        self.top_scroll = Qt.QScrollArea()
        self.top_scroll.setFrameStyle(Qt.QFrame.NoFrame)
        self.top_scroll_layout.addWidget(self.top_scroll)
        self.top_scroll.setWidgetResizable(True)
        self.top_widget = Qt.QWidget()
        self.top_scroll.setWidget(self.top_widget)
        self.top_layout = Qt.QVBoxLayout(self.top_widget)
        self.top_grid_layout = Qt.QGridLayout()
        self.top_layout.addLayout(self.top_grid_layout)

        self.settings = Qt.QSettings("GNU Radio", "test")

        try:
            if StrictVersion(Qt.qVersion()) < StrictVersion("5.0.0"):
                self.restoreGeometry(self.settings.value("geometry").toByteArray())
            else:
                self.restoreGeometry(self.settings.value("geometry"))
        except:
            pass

        ##################################################
        # Variables
        ##################################################
        self.samp_rate = samp_rate = 32000
        self.frequency = frequency = 746e6

        ##################################################
        # Blocks
        ##################################################
        self._frequency_range = Range(742e6, 755e6, 1, 746e6, 200)
        self._frequency_win = RangeWidget(self._frequency_range, self.set_frequency, 'frequency', "counter_slider", float)
        self.top_grid_layout.addWidget(self._frequency_win)
        self.osmosdr_sink_0 = osmosdr.sink(
            args="numchan=" + str(1) + " " + ""
        )
        self.osmosdr_sink_0.set_time_unknown_pps(osmosdr.time_spec_t())
        self.osmosdr_sink_0.set_sample_rate(samp_rate)
        self.osmosdr_sink_0.set_center_freq(frequency, 0)
        self.osmosdr_sink_0.set_freq_corr(0, 0)
        self.osmosdr_sink_0.set_gain(2, 0)
        self.osmosdr_sink_0.set_if_gain(20, 0)
        self.osmosdr_sink_0.set_bb_gain(20, 0)
        self.osmosdr_sink_0.set_antenna('', 0)
        self.osmosdr_sink_0.set_bandwidth(0, 0)
        self.analog_sig_source_x_0 = analog.sig_source_c(samp_rate, analog.GR_SIN_WAVE, frequency, 1, 0, 0)

        ##################################################
        # Connections
        ##################################################
        self.connect((self.analog_sig_source_x_0, 0), (self.osmosdr_sink_0, 0))

    def closeEvent(self, event):
        self.settings = Qt.QSettings("GNU Radio", "test")
        self.settings.setValue("geometry", self.saveGeometry())
        event.accept()

    def setStyleSheetFromFile(self, filename):
        try:
            if not os.path.exists(filename):
                filename = os.path.join(
                    gr.prefix(), "share", "gnuradio", "themes", filename)
            with open(filename) as ss:
                self.setStyleSheet(ss.read())
        except Exception as e:
            print(e, file=sys.stderr)

    def get_samp_rate(self):
        return self.samp_rate

    def set_samp_rate(self, samp_rate):
        self.samp_rate = samp_rate
        self.analog_sig_source_x_0.set_sampling_freq(self.samp_rate)
        self.osmosdr_sink_0.set_sample_rate(self.samp_rate)

    def get_frequency(self):
        return self.frequency

    def set_frequency(self, frequency):
        self.frequency = frequency
        self.analog_sig_source_x_0.set_frequency(self.frequency)
        self.osmosdr_sink_0.set_center_freq(self.frequency, 0)

def main(top_block_cls=test, options=None):

    if StrictVersion("4.5.0") <= StrictVersion(Qt.qVersion()) < StrictVersion("5.0.0"):
        style = gr.prefs().get_string('qtgui', 'style', 'raster')
        Qt.QApplication.setGraphicsSystem(style)
    qapp = Qt.QApplication(sys.argv)

    tb = top_block_cls()
    tb.start()
    tb.setStyleSheetFromFile("/Applications/GNURadio.app/Contents/MacOS/usr/qml/Qt/WebSockets/qmldir")
    tb.show()

    def sig_handler(sig=None, frame=None):
        Qt.QApplication.quit()

    signal.signal(signal.SIGINT, sig_handler)
    signal.signal(signal.SIGTERM, sig_handler)

    timer = Qt.QTimer()
    timer.start(500)
    timer.timeout.connect(lambda: None)

    def quitting():
        tb.stop()
        tb.wait()
    qapp.aboutToQuit.connect(quitting)
    qapp.exec_()

if __name__ == '__main__':
    main()

### Expected behaviour
I used to get .013v ac from volt metter no mater what frequency it ran

gose egg nothing no voltage

### Version information
**Operating system**:
macos
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

markl17, 2020-11-19:
hi can someone tell me which chip I fried because everything else about the device seems to work except its not putting voltage out and  how do I increase the voltage is it the db thing and which is I am not using it for what everyone else is using it
i just need inexpensive signal generator so I am not well versed in db before signal is amplified and after signal is amplified and hormonics and shadows of the signal began to appear just need something like 2.1 ghz or 2100000000 hz . another question can it du precision like 2100000000.123456789 hz

straithe, 2021-01-26:
Are you still experiencing this issue?

markl17, 2021-01-26:
> Do you still need assistance with this issue?

yes which pins and how do I trouble shoot it

mossmann, 2021-02-04:
AC voltmeters are typically designed for AC mains frequencies of 50-60 Hz. HackRF One produces frequencies of 1 MHz to 6 GHz. It is normal for an AC voltmeter to give no valid reading of the HackRF One output.

I notice that your flowgraph configures the HackRF One with a sample rate of 32000 samples per second. The lowest supported sample rate is 2 million (2e6) samples per second. Use a sample rate between 2e6 and 20e6 to ensure that the HackRF One functions correctly.

You might prefer to use the `hackrf_transfer` command line utility to generate test signals. For example, to generate a signal at 746 MHz, type `hackrf_transfer -c 127 -a 0 -x 47 -f 746000000`. See `hackrf_transfer -h` for information about the command line options.

markl17, 2021-02-04:
that is not the issue when I connect freuency counter to hackrf I get
749.993 mhz when hackrf said it produces 750 mhz which is close enough the
truth is between them but when I attach one of these

Jun 10, 2020
ORDER NUMBER
114149664477-2088416880001
SOLD BY
2019happyday
<http://www.ebay.com/usr/2019happyday?_trksid=p2057872.m2749.l2754>(1 item)
*Order 10M-3GHZ 120DB large dynamic range CNC gain amplifier 0.01DB step RF
amplifier*View order details
<https://www.ebay.com/vod/FetchOrderDetails?itemid=114149664477&transId=2088416880001&_trksid=p2057872.m2749.l2673>
[image: 10M-3GHZ 120DB large dynamic range CNC gain amplifier 0.01DB step
RF amplifier]
10M-3GHZ 120DB large dynamic range CNC gain amplifier 0.01DB step RF
amplifier
( 114149664477 )

the counter still used to tell me more or less the same and even after I
connected this
[image: LNA Low Noise 50K-4G High Gain 25DB 0.8G High Gain Flatness RF
Amplifier New。]
LNA Low Noise 50K-4G High Gain 25DB 0.8G High Gain Flatness RF Amplifier
New。
( 163976343816 )

qit still told me the same but than something happened and both of these
amplifiers says who knows what but hackrf by itself still says the write
thing to the counter

mossmann, 2021-02-04:
Neither of those eBay links works for me. I am not understanding your problem.

mossmann, 2021-03-02:
I'm closing this due to lack of response. If you would still like help, please open a new issue describing the problem clearly and in a way that can help us try to reproduce it.

closed 2021-03-02 by mossmann

markl17, 2021-06-18:
I have found out that my amp chip u21 has died do you know what is the name
of the chip in question so I can replace it

gbuzogany, 2021-06-18:
I believe U21 is the TPS62410.
[image]

markl17, 2021-06-18:
this picture shows a a six legs 3 on each the the part you are showing has
8 4 on each side
are you sure

straithe, 2021-06-18:
This conversation continues in GitHub issue #901

roadmutt, 2021-06-19:
i want to say its the one with 6 legs just above the tiny ceramic resistor
just above the right hand side of the micro USB port im new to this
magnitude of electrical engineering so to speak if you tell me how to ill
test it if possible i dont have a logic probe but i have a good
voltage/combo meter

thanks for the help
Kevyn Smith

markl17, 2021-06-20:
I think I was talking about the amplifier chip there are two of them
they’re located right  around antenna area behind the metal shield if
measure resistans between the third leg and couple of other you will see
some values but in a dead one it will be blank, this is the preamp chip and
amp chip the provide the 14 dB of signal and 40db from -70db where the
signal starts so if everything is working than you should see about .014v
of ac coming out if not it will be 0 a big guss egg like I have now