greatscottgadgets/hackrf #808: osmocom Sink causes runtime error when generating in GNU radio 3.8

melfvelin, 2020-11-19:
### Steps to reproduce
1. Fresh install of gnuradio 3.8 + hackrf (including prerequisits) on Ubuntu 20.04
2. Try generating a simple tx flowgraph with osmocom Sink
3.

### Expected behaviour
Succesful generation

### Actual behaviour
Error message see below

>>> Done (return code 1)

### Version information
**Operating system**:
Ubuntu 20.04.1 LTS
64-bit
**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000057b068dc21475363
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005d474d

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GNU Radio 3.8.1.0
### Output
```

Generating: '/home/kiruna/B200_TX_TEST.py'
Generate Error: (NameError("'dc_offset_mode8' is not defined"), 'osmosdr.sink(\n    args="numchan=" + str(${nchan}) + " " + ${args}\n)\n% if context.get(\'num_mboards\')() > 0:\n% if context.get(\'clock_source0\')():\nself.${id}.set_clock_source(${clock_source0}, 0)\n% endif\n% if context.get(\'time_source0\')():\nself.${id}.set_time_source(${time_source0}, 0)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 1:\n% if context.get(\'clock_source1\')():\nself.${id}.set_clock_source(${clock_source1}, 1)\n% endif\n% if context.get(\'time_source1\')():\nself.${id}.set_time_source(${time_source1}, 1)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 2:\n% if context.get(\'clock_source2\')():\nself.${id}.set_clock_source(${clock_source2}, 2)\n% endif\n% if context.get(\'time_source2\')():\nself.${id}.set_time_source(${time_source2}, 2)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 3:\n% if context.get(\'clock_source3\')():\nself.${id}.set_clock_source(${clock_source3}, 3)\n% endif\n% if context.get(\'time_source3\')():\nself.${id}.set_time_source(${time_source3}, 3)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 4:\n% if context.get(\'clock_source4\')():\nself.${id}.set_clock_source(${clock_source4}, 4)\n% endif\n% if context.get(\'time_source4\')():\nself.${id}.set_time_source(${time_source4}, 4)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 5:\n% if context.get(\'clock_source5\')():\nself.${id}.set_clock_source(${clock_source5}, 5)\n% endif\n% if context.get(\'time_source5\')():\nself.${id}.set_time_source(${time_source5}, 5)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 6:\n% if context.get(\'clock_source6\')():\nself.${id}.set_clock_source(${clock_source6}, 6)\n% endif\n% if context.get(\'time_source6\')():\nself.${id}.set_time_source(${time_source6}, 6)\n% endif\n% endif\n% if context.get(\'num_mboards\')() > 7:\n% if context.get(\'clock_source7\')():\nself.${id}.set_clock_source(${clock_source7}, 7)\n% endif\n% if context.get(\'time_source7\')():\nself.${id}.set_time_source(${time_source7}, 7)\n% endif\n% endif\n% if sync == \'sync\':\nself.${id}.set_time_unknown_pps(osmosdr.time_spec_t())\n% elif sync == \'pc_clock\':\nself.${id}.set_time_now(osmosdr.time_spec_t(time.time()), osmosdr.ALL_MBOARDS)\n% endif\nself.${id}.set_sample_rate(${sample_rate})\n% if context.get(\'nchan\')() > 0:\nself.${id}.set_center_freq(${freq0}, 0)\nself.${id}.set_freq_corr(${corr0}, 0)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode0}, 0)\nself.${id}.set_iq_balance_mode(${iq_balance_mode0}, 0)\nself.${id}.set_gain_mode(${gain_mode0}, 0)\n% endif\nself.${id}.set_gain(${gain0}, 0)\nself.${id}.set_if_gain(${if_gain0}, 0)\nself.${id}.set_bb_gain(${bb_gain0}, 0)\nself.${id}.set_antenna(${ant0}, 0)\nself.${id}.set_bandwidth(${bw0}, 0)\n% endif\n% if context.get(\'nchan\')() > 1:\nself.${id}.set_center_freq(${freq1}, 1)\nself.${id}.set_freq_corr(${corr1}, 1)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode1}, 1)\nself.${id}.set_iq_balance_mode(${iq_balance_mode1}, 1)\nself.${id}.set_gain_mode(${gain_mode1}, 1)\n% endif\nself.${id}.set_gain(${gain1}, 1)\nself.${id}.set_if_gain(${if_gain1}, 1)\nself.${id}.set_bb_gain(${bb_gain1}, 1)\nself.${id}.set_antenna(${ant1}, 1)\nself.${id}.set_bandwidth(${bw1}, 1)\n% endif\n% if context.get(\'nchan\')() > 2:\nself.${id}.set_center_freq(${freq2}, 2)\nself.${id}.set_freq_corr(${corr2}, 2)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode2}, 2)\nself.${id}.set_iq_balance_mode(${iq_balance_mode2}, 2)\nself.${id}.set_gain_mode(${gain_mode2}, 2)\n% endif\nself.${id}.set_gain(${gain2}, 2)\nself.${id}.set_if_gain(${if_gain2}, 2)\nself.${id}.set_bb_gain(${bb_gain2}, 2)\nself.${id}.set_antenna(${ant2}, 2)\nself.${id}.set_bandwidth(${bw2}, 2)\n% endif\n% if context.get(\'nchan\')() > 3:\nself.${id}.set_center_freq(${freq3}, 3)\nself.${id}.set_freq_corr(${corr3}, 3)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode3}, 3)\nself.${id}.set_iq_balance_mode(${iq_balance_mode3}, 3)\nself.${id}.set_gain_mode(${gain_mode3}, 3)\n% endif\nself.${id}.set_gain(${gain3}, 3)\nself.${id}.set_if_gain(${if_gain3}, 3)\nself.${id}.set_bb_gain(${bb_gain3}, 3)\nself.${id}.set_antenna(${ant3}, 3)\nself.${id}.set_bandwidth(${bw3}, 3)\n% endif\n% if context.get(\'nchan\')() > 4:\nself.${id}.set_center_freq(${freq4}, 4)\nself.${id}.set_freq_corr(${corr4}, 4)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode4}, 4)\nself.${id}.set_iq_balance_mode(${iq_balance_mode4}, 4)\nself.${id}.set_gain_mode(${gain_mode4}, 4)\n% endif\nself.${id}.set_gain(${gain4}, 4)\nself.${id}.set_if_gain(${if_gain4}, 4)\nself.${id}.set_bb_gain(${bb_gain4}, 4)\nself.${id}.set_antenna(${ant4}, 4)\nself.${id}.set_bandwidth(${bw4}, 4)\n% endif\n% if context.get(\'nchan\')() > 5:\nself.${id}.set_center_freq(${freq5}, 5)\nself.${id}.set_freq_corr(${corr5}, 5)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode5}, 5)\nself.${id}.set_iq_balance_mode(${iq_balance_mode5}, 5)\nself.${id}.set_gain_mode(${gain_mode5}, 5)\n% endif\nself.${id}.set_gain(${gain5}, 5)\nself.${id}.set_if_gain(${if_gain5}, 5)\nself.${id}.set_bb_gain(${bb_gain5}, 5)\nself.${id}.set_antenna(${ant5}, 5)\nself.${id}.set_bandwidth(${bw5}, 5)\n% endif\n% if context.get(\'nchan\')() > 6:\nself.${id}.set_center_freq(${freq6}, 6)\nself.${id}.set_freq_corr(${corr6}, 6)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode6}, 6)\nself.${id}.set_iq_balance_mode(${iq_balance_mode6}, 6)\nself.${id}.set_gain_mode(${gain_mode6}, 6)\n% endif\nself.${id}.set_gain(${gain6}, 6)\nself.${id}.set_if_gain(${if_gain6}, 6)\nself.${id}.set_bb_gain(${bb_gain6}, 6)\nself.${id}.set_antenna(${ant6}, 6)\nself.${id}.set_bandwidth(${bw6}, 6)\n% endif\n% if context.get(\'nchan\')() > 7:\nself.${id}.set_center_freq(${freq7}, 7)\nself.${id}.set_freq_corr(${corr7}, 7)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode7}, 7)\nself.${id}.set_iq_balance_mode(${iq_balance_mode7}, 7)\nself.${id}.set_gain_mode(${gain_mode7}, 7)\n% endif\nself.${id}.set_gain(${gain7}, 7)\nself.${id}.set_if_gain(${if_gain7}, 7)\nself.${id}.set_bb_gain(${bb_gain7}, 7)\nself.${id}.set_antenna(${ant7}, 7)\nself.${id}.set_bandwidth(${bw7}, 7)\n% endif\n% if context.get(\'nchan\')() > 8:\nself.${id}.set_center_freq(${freq8}, 8)\nself.${id}.set_freq_corr(${corr8}, 8)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode8}, 8)\nself.${id}.set_iq_balance_mode(${iq_balance_mode8}, 8)\nself.${id}.set_gain_mode(${gain_mode8}, 8)\n% endif\nself.${id}.set_gain(${gain8}, 8)\nself.${id}.set_if_gain(${if_gain8}, 8)\nself.${id}.set_bb_gain(${bb_gain8}, 8)\nself.${id}.set_antenna(${ant8}, 8)\nself.${id}.set_bandwidth(${bw8}, 8)\n% endif\n% if context.get(\'nchan\')() > 9:\nself.${id}.set_center_freq(${freq9}, 9)\nself.${id}.set_freq_corr(${corr9}, 9)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode9}, 9)\nself.${id}.set_iq_balance_mode(${iq_balance_mode9}, 9)\nself.${id}.set_gain_mode(${gain_mode9}, 9)\n% endif\nself.${id}.set_gain(${gain9}, 9)\nself.${id}.set_if_gain(${if_gain9}, 9)\nself.${id}.set_bb_gain(${bb_gain9}, 9)\nself.${id}.set_antenna(${ant9}, 9)\nself.${id}.set_bandwidth(${bw9}, 9)\n% endif\n% if context.get(\'nchan\')() > 10:\nself.${id}.set_center_freq(${freq10}, 10)\nself.${id}.set_freq_corr(${corr10}, 10)\n% if context.get(\' [... cut ...] freq(${freq31}, 31)\nself.${id}.set_freq_corr(${corr31}, 31)\n% if context.get(\'sourk\') == \'source\':\nself.${id}.set_dc_offset_mode(${dc_offset_mode31}, 31)\nself.${id}.set_iq_balance_mode(${iq_balance_mode31}, 31)\nself.${id}.set_gain_mode(${gain_mode31}, 31)\n% endif\nself.${id}.set_gain(${gain31}, 31)\nself.${id}.set_if_gain(${if_gain31}, 31)\nself.${id}.set_bb_gain(${bb_gain31}, 31)\nself.${id}.set_antenna(${ant31}, 31)\nself.${id}.set_bandwidth(${bw31}, 31)\n% endif\n')
>>> Failure

```

miek, 2020-11-19:
Are you using the latest version of gr-osmosdr? This should be fixed in v0.2.2

melfvelin, 2020-11-20:
gr-osmosdr/focal,now 0.2.0-2 amd64 [installed]
Installed with apt-get

miek, 2021-01-04:
Sorry, I must've missed the response before. So yeah, that's v0.2.0 which has this bug. You'll need to get v0.2.2 by either building from source, from a PPA, or I think it's available in Ubuntu 20.10 now.

closed 2021-01-04 by miek