greatscottgadgets/hackrf #811: Problems with static noise only when plugged into a PC

HOAXsk8, 2020-12-12:
### Steps to reproduce
1. Plug in the HackRF One with portapack
2. Open HDSDR v 2.8.0
3. Tune to FM radio station

### Expected behaviour
I am expecting to hear a clear radio station. If I use another receiver with the same settings it works. I have tried with GQRX and GNU Radio Companion. It is a problem with the HackRF One.

### Actual behaviour
Fuzzy static noise with a hint of audio from the station occasionally.

### Version information
Windows 10 and ParrotOS:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

HOAXsk8, 2020-12-12:
I forgot to mention that if I use the HackRF One as a standalone device I can tune in to the radio with no issues.

straithe, 2021-01-26:
Have you tried switching out the cable that you used to connect the HackRF to the computer? Does the same thing happen when you use software other than HDSDR v 2.8.0 with the HackRF?

HOAXsk8, 2021-01-26:
Hi I didn't try switching the cable. I will try it. I'm certain it is to do with the power supply from my laptop. When I use it standalone with a power bank there is no problem. I will try another cable later today. I also tried it on another laptop with the same cable and had the same issue. So maybe it is the cable.

straithe, 2021-02-11:
Have you had a chance to try the HackRF with a new cable?

closed 2021-03-18 by straithe