greatscottgadgets/hackrf #813: How to start YardStick One ???

matriarch-007, 2020-12-22:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

mvaden78, 2021-01-19:
Unrelated question .sorry.
any chance the small holes on the yard stick can plug into  my hackRF to run it at full duplex rather then half.

straithe, 2021-01-24:
> Unrelated question .sorry.
> any chance the small holes on the yard stick can plug into my hackRF to run it at full duplex rather then half.

No. These devices are not designed to plug into each other.

straithe, 2021-01-24:
> ### Steps to reproduce
> 
> 
> ### Expected behaviour
> 
> Tell us what you expect should happen
> ### Actual behaviour
> 
> Tell us what happens instead
> ### Version information
> 
> **Operating system**:
> 
> **hackrf_info output:**
> 
> If you are reporting a problem that involves third party software
> (GNU Radio, Gqrx, etc), please report the version here.
> ### Output
> 
> ```
> Insert any commandline or build output here
> ```

Please check out the [RFCat Documentation](https://github.com/atlas0fd00m/rfcat). If you have further YARD Stick One questions please ask them on the [YARD Stick One Repo](https://github.com/greatscottgadgets/yardstick).

closed 2021-01-24 by straithe