greatscottgadgets/hackrf #815: Question on the Hardware

Redjack1996, 2020-12-30:
### MAX2837
Hello, I'm planning on starting an open source hardware project for an SDR and I've been looking at the HackRF schematics for some inspiration. 

I assume the RFFC5072 device is used for down conversion and the MAX2837 is used for IF to BB and IQ demod. 

I have a query about the MAX2837 IC, that chip is specced for an RF input freq. 2.3 to 2.7Ghz, but HackRF operates down to 1MHz. When operating at this low frequency do you use the RFFC5072 mixers to upconvert to the input freq. range of the MAX device or is the MAX device okay at these frequencies, or am I missing something and being a bit silly?

Many thanks

schneider42, 2021-01-04:
The RFFC5072 is used for upconversion in this case. The IF is always kept in the range mentioned above.

Redjack1996, 2021-01-04:
Thanks very much!

closed 2021-01-23 by mossmann