greatscottgadgets/hackrf #816: problem with spiflash

superchuckles, 2021-01-03:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
ok, i'm having the same issue as above, with one differeence.  when i run 'hackrf_spiflash -w hackrf_one_usb.bin' it does the folloowing:

File size 22024 bytes.
Erasing SPI flash.
Writing 22024 bytes at 0x000000
and stops.  i can leave it all day & when i come back it's still at that same point.

miek, 2021-01-04:
> ok, i'm having the same issue as above, with one differeence. 
I don't see the other issue you're referring to, did you mean to paste a link or reply to another thread?

Could you try running it again with the -v and -s flags and paste the output? ie: `hackrf_spiflash -v -s -w hackrf_one_usb.bin`

mossmann, 2021-01-23:
Are you still having trouble?  We will close this issue if we don't get a reply within the next month?

closed 2021-02-23 by straithe