greatscottgadgets/hackrf #822: Switching antenna with hackrf Gpio

waqarulhaq129, 2021-01-28:
Can anyone explain how to access hackrf gpio? I need to switch antenna with hackrf clock for direction finding as explained by balint seeber in "huff-duff". I have to write 2 Gpio pins to output 1 and 0 according to frequency i tell hackrf.

mossmann, 2021-02-13:
GPIO can be accessed from firmware. It is not exposed to host software. I'm not sure what shape it is in, but @docprofsky implemented pseudo-doppler antenna switching using the SCTimer peripheral in HackRF One's LPC4320. You can find it in [this branch](https://github.com/docprofsky/hackrf/tree/operacake-sctimer). You may also be interested in this talk that we gave at ShmooCon on the subject: https://archive.org/details/Shmoocon2018/Shmoocon2018-Pseudo-dopplerRedux.mp4

closed 2021-02-13 by mossmann

docprofsky, 2021-02-16:
Looking at the SCTimer code, it looks like the firmware side is complete. However, the host configuration code is not. I will work on the software with the hope of getting it finished in about a fortnight.

nikstp, 2022-09-26:
Has there been any progress on this? Is this possible without the Opera Cake board?

martinling, 2022-09-26:
Opera Cake is our recommended way to do antenna switching, it's in production and just starting to be available from distributors.

If you want to drive your own switching solution from GPIO then the simplest way to do that is going to be to adapt the code that's already there for driving Opera Cake.

nikstp, 2022-09-26:
@martinling thank you for your reply! Yeah, using the Opera Cake would be fine but as it's not out and I'm looking for a solution right now, I might have to do with trying to use GPIO. Do you have any estimates on when Opera Cake will be available thru US-based distributors?

martinling, 2022-09-26:
They're shipping out as we speak. I know some distributors received theris today but I'm not sure off-hand which ones. It looks like [Hacker Warehouse](https://hackerwarehouse.com/product/opera-cake-for-hackrf/) may have them now. If you need more info on availability please email info@greatscottgadgets.com.