greatscottgadgets/hackrf #823: Problem with osmocom - HackRF with GNU Radio 3.9.0

VasilyRakche, 2021-01-28:
### Steps to reproduce
1. Install GNURadio 3.9.0 and all dependencies from source according to the explanation in this [issue](https://github.com/gnuradio/gnuradio/issues/3918)
2. Install gr-osmocom from source according to [this](https://osmocom.org/projects/gr-osmosdr/wiki)
3. Install hackrf - just with sudo apt-get install hackrf 
4. Change udev rules so the hackrf_info works properly according to [HackRF FAQ](https://github.com/mossmann/hackrf/wiki/FAQ)

### Expected behaviour
Running simple script with osmocom source and frequency sink.

### Actual behaviour
Getting the error that osmocom cant find the device (I was previously working with the GNURadio 3.8, everything was working fine).

### Version information
**Operating system**: Ubuntu 20.04

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000087c867dc2a852b5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0078475f
```
GNU Radio version 3.9.0.0

### Output
```
<<< Welcome to GNU Radio Companion 3.9.0.0 >>>

Block paths:
	/home/vasko/opt/gnuradio-3.9.0git/share/gnuradio/grc/blocks

Loading: "/home/vasko/UNITN1_/dncs_labs/gnuradio/wifi_discover.grc"
>>> Done

Loading: "/home/vasko/UNITN1_/dncs_labs/gnuradio/tryout.grc"
>>> Done

Generating: '/home/vasko/UNITN1_/dncs_labs/gnuradio/top_block.py'
>>> Warning: This flow graph contains a throttle block and another rate limiting block, e.g. a hardware source or sink. This is usually undesired. Consider removing the throttle block.

Executing: /usr/bin/python3 -u /home/vasko/UNITN1_/dncs_labs/gnuradio/top_block.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.9.0.0
built-in source types: file rtl_tcp uhd rfspace redpitaya 
[INFO] [UHD] linux; GNU C++ version 9.3.0; Boost_107100; UHD_3.15.0.tmp-0-gaea0e2de
Traceback (most recent call last):
  File "/home/vasko/UNITN1_/dncs_labs/gnuradio/top_block.py", line 289, in <module>
    main()
  File "/home/vasko/UNITN1_/dncs_labs/gnuradio/top_block.py", line 267, in main
    tb = top_block_cls()
  File "/home/vasko/UNITN1_/dncs_labs/gnuradio/top_block.py", line 179, in __init__
    self.osmosdr_source_0 = osmosdr.source(
RuntimeError: No supported devices found (check the connection and/or udev rules).

>>> Done (return code 1)
```

miek, 2021-02-01:
It looks like gr-osmosdr was not built with HackRF support enabled as `hackrf` is not listed in `built-in source types: file rtl_tcp uhd rfspace redpitaya `. You need to install HackRF before building gr-osmosdr from source.

closed 2021-02-11 by miek

T3zcatlip0ca, 2022-06-17:
Hello.
I tried solved the same problem, my hackRf present the correct output for the comand hackrf_info in ubuntu terminal, but when i try run a flow chart in GNU radio presents the problem:

 File "/home/cp/Documentos/FM_Receiver.py", line 407, in <module>
    main()
  File "/home/cp/Documentos/FM_Receiver.py", line 385, in main
    tb = top_block_cls()
  File "/home/cp/Documentos/FM_Receiver.py", line 255, in __init__
    self.osmosdr_source_0 = osmosdr.source(
RuntimeError: No supported devices found (check the connection and/or udev rules).

I tried modify the udev rules but but still not working.

Sorry, were you able to solve the problem?

Thanks for your attention

straithe, 2022-06-17:
Please open a new issue and we will work with you to help get your HackRF working.

T3zcatlip0ca, 2022-06-17:
ok, perfect