greatscottgadgets/hackrf #824: USB Filter

James-Brown, 2021-01-28:
I have a lot of noise coming down my HackRF USB cable. The noise is generated by some DC motors that move my antenna. Does anyone have some experience with USB/USB isolators with noise filters built-in?

draeath, 2021-03-15:
Ensure your cable is shielded - a quick but not always 100% accurate way to test this is to ensure the external metal parts of the USB connector on both ends have electrical continuity. (the unreliability is due to the fact this doesn't guarantee this grounding is a proper shield woven down the cable or just an extra wire). Additionally you're going to want a cable that has a [ferrite choke](https://en.wikipedia.org/wiki/Ferrite_bead) integral (or you could add one yourself!).

Is there any way you can get the motors on a different circuit / ground plane?

Additionally if you have not yet installed one, it may be worth installing the [RF shield](https://www.nooelec.com/store/sdr/sdr-addons/hackrf-rf-shield.html) in your hackrf in case the noise is not actually coming through the USB cable. Note this will require some advanced soldering skills.

straithe, 2021-04-23:
@James-Brown did @draeath's suggestions help or do you still need assistance?