greatscottgadgets/hackrf #825: Q: Inner hardware explaination

blackwiz4rd, 2021-01-30:
Hi,
is there any detailed documentation of the inner hardware? I do not think the wiki provides enough details and explaination:
https://github.com/mossmann/hackrf/wiki/Hardware-Components
Thanks a lot!

miek, 2021-02-04:
I don't know of any other documentation. Can you expand on what details you'd like to see, or any specific questions you have?

blackwiz4rd, 2021-02-04:
Sure, for example RTL-SDR have very detailed explanation of the inner hardware as in this blog post: `https://web.archive.org/web/20191120002326/http://ajoo.blog/intro-to-rtl-sdr-part-i-principles-and-hardware.html`. I would like to know if there is anything similar for the hackrf. That would be useful to understand how the device works.
Thanks,
Luca.

mossmann, 2021-02-04:
The complete hardware design files (for KiCad) are here: https://github.com/mossmann/hackrf/tree/master/hardware/hackrf-one

You can find an explanation of the HackRF hardware architecture in this video from the 2012 GNU Radio Conference (It is about Jawbreaker which had the same architecture as HackRF One.): https://www.youtube.com/watch?v=wzu7lXOEl4k

blackwiz4rd, 2021-02-08:
> The complete hardware design files (for KiCad) are here: https://github.com/mossmann/hackrf/tree/master/hardware/hackrf-one
>
> You can find an explanation of the HackRF hardware architecture in this [...]

Thanks for your answer. In the video you talk about bypassing the wideband frontend, would that be helpful in case of acquiring a better signal at hf frequency (13.56MHz, NFC)? If so, can it be done with the already implemented hackrf tools? Do you have any suggestion to acquire signal at this frequency with hackrf?
Thanks a lot.

mossmann, 2021-02-08:
> Thanks for your answer. In the video you talk about bypassing the wideband frontend, would that be helpful in case of acquiring a better signal at hf frequency (13.56MHz, NFC)? If so, can it be done [...]

Bypassing the wideband front-end allows operation between 2.15 and 2.75 GHz (the IF). It does not allow operation at 13.56 MHz. It is possible to tap into the analog baseband signals directly (header P9) for operation at low frequencies such as 13.56 MHz, but that would require some external interface circuitry.

Passive monitoring of NFC signals is typically quite easy at close range with a HackRF One and an appropriate loop antenna. In fact, the signals can be so strong that I advise using a 20 dB attenuator between the antenna and the HackRF One. Transmitting NFC signals is another matter entirely; the near field load modulation used by NFC is totally different than radio propagation. For that you need a specialized device such as a ChameleonMini or Proxmark.

blackwiz4rd, 2021-02-15:
> Bypassing the wideband front-end allows operation between 2.15 and 2.75 GHz (the IF). It does not allow operation at 13.56 MHz. It is possible to tap into the analog baseband signals directly [...]

Thank you very much for your answer

closed 2021-02-15 by blackwiz4rd