greatscottgadgets/hackrf #826: Hardware defect or software bug, non-existent spike in spectrum graph with HackRF One

noviceFedora, 2021-01-30:
### Steps to reproduce
1. Open SDR Console or GQRX
2. Select HackRF One
3. Start playing

### Expected behaviour
Spectrum graph should change depending on the rf noise around the antenna.

### Actual behaviour
There is always a spike in the center of the spectrum graph, no matter which frequencies I'm scanning

### Version information
**Operating system**:

Dragon OS
Windows 10 20H2 V2

**hackrf_info output:** not available

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here. 

SDR Console 3.0.26

### Output
```
Insert any commandline or build output here
```
You can check the screenshot on this link: https://i.ibb.co/n64h104/hackrf.png

I'm referring to the spike in the center at 847.500 in the screenshot, I've scanned the entire frequency range, and there is always that spike no matter what frequencies I'm scanning, I've checked this in Dragon OS and Windows 10 with SDR Console, HDSDR, GQRX, CubicSDR, etc and there is this spike in all of them no matter the frequency. 

When I check that place with AM, FM, etc the signal strength is increasing but there is no noise or anything, the waterfall is also showing the same activity there.

In GQRX, I can drag the red scan line towards the right, so if the center frequency was at 900MHz and that spike was there, if I drag the red scan to the right, the 900MHz would towards the left but that spike remains in that same place with that same activity, if I move the red scan line to the 900MHz on it's new position at left, I don't any such activity. 

Is this a hardware defect or software bug?

miek, 2021-01-30:
That spike is normal, there's a good explanation of it here: https://github.com/mossmann/hackrf/wiki/FAQ#what-is-this-big-spike-in-the-center-of-my-received-spectrum

noviceFedora, 2021-01-31:
> That spike is normal, there's a good explanation of it here: https://github.com/mossmann/hackrf/wiki/FAQ#what-is-this-big-spike-in-the-center-of-my-received-spectrum

How can I verify that the reason you are giving is the real cause? How can I know this is not disinformation, because I suspect there is an implant in me and I bought HackRF One to find the frequency it receives, to throw me off course on this, someone could have hacked it to show phantom noise so that I'll never be able to find the real frequency. 

I have read the question on the FAQ. Is HackRF One different than rtlsdr? Because I don't see such phantom noise on rtlsdr.

Hoernchen, 2021-01-31:
"The rtlsdr" dc offset varies because it's actually not really one single device due to multiple tuners and hardware versions, so comparing dc offsets is difficult, and more expensive transceiver like the ones sold by ADI obviously try to get rid of it. I suppose you could just search for "dc offset" in the rtlsdr subreddit, the reason why there is a faq entry is becausee this is the questions everyone new to sdr has been asking for the past 8 years.

That being said I don't see how you could verify anything beyond what the faq explains, considering that I'd just ensure you never see the signal you're looking for by adjusting the software you're using - but I'd suppress the dc offset as well to get around the usual dc offset question everyone asks...

closed 2021-02-01 by miek