greatscottgadgets/hackrf #827: HackRF One Spikes only and noise (PCB 13 February 2014)

yourSpace24, 2021-02-08:
### Steps to reproduce
Using GQRX Version 2.11.5 to tune into some FM Radio signal got a channel here on 100mhz if i try to tune in i see the spike and noise nothing more.

### Expected behaviour
from the examples you should get some audio

### Actual behaviour
only noise on audio. 
[image]
[image]

### Version information
**Operating system**:
Debian 10

**hackrf_info output:**
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000017c467dc2d1b47c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0061475e

**Operating system**:
If you are reporting a problem that involves third party software 
GQRX

### Output
```
Insert any commandline or build output here
```

straithe, 2021-02-13:
I think you might be experiencing an issue with your antenna. If you can, please try swapping your antenna out for another. If you don't have an antenna you can use a straightened paperclip.

yourSpace24, 2021-02-14:
I did and used a paperclip no change looks always the same. Is there maybe the amplifier fried ? the device was new unpacked it and just turned it on.

straithe, 2021-02-15:
Can you send me two new screenshots, one where in the control tab of GQRX the RF is set to 0 and another where in the control tab of GQRX the RF is set to 14?

yourSpace24, 2021-02-16:
sure
[image]
[image]

straithe, 2021-02-16:
Are you able to pick up FM radio stations on any other radio receivers you may have?

yourSpace24, 2021-02-16:
from my car radio just 5meter away yes.

straithe, 2021-02-18:
Thank you for the extra information. It does look like you have a hardware failure. Please get in contact with the reseller that sold you your HackRF One and they will replace it for you.

closed 2021-02-18 by straithe