greatscottgadgets/hackrf #828: hackrf_start_rx_sweep() failed:

reinhs, 2021-02-10:
### Steps to reproduce
1. Clean all hackrf installs and libs from system or start with new system
2. git clone https://github.com/mossmann/hackrf
2. follow compile steps on host readme
3. execute hackrf_sweep

### Expected behavior
Sweeps start streaming out until ctrl-c

### Actual behavior
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)
Usage:
...

### Version information
**Operating system**:
Ubuntu 18.04 and also Pi3

**hackrf_info output:**
hackrf_info version: git-f202160
libhackrf version: git-f202160 (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000017c467dc2969a1c3
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00414f5c

----
If I do a "sudo apt install hackrf" and do a sweep it works perfect so my board/firmware/cables are fine. I just can't make a compiled version work. If I install something like QSpectrumAnalyzer it works without issue. What am I missing on the clone/compile/install? Also tried 2018.1.1 firmware with same result.

reinhs, 2021-02-10:
Ah, FWIW, if I download the 2017.02.1 release and build it works perfectly. I guess cloning the main repo doesn't work with the older firmware? If so, my fault. Sorry.

miek, 2021-02-10:
The latest version of the host software requires firmware built from git too, as there were some new API calls introduced since the last release - that's why it throws that error.

reinhs, 2021-02-10:
Perfect! Thank you Sir, totally makes sense now.

closed 2021-02-10 by reinhs