greatscottgadgets/hackrf #829: hackrf_sweep: hackrf_start_rx_sweep() failed: Pipe error (-1000)

frinkemon, 2021-02-11:
### Steps to reproduce

1. Compile current release on Ubuntu 20.0.4

2. Run hackrf_sweep:
3. 
leigh@boxy:~$ /usr/local/bin/hackrf_info 
hackrf_info version: git-f202160
libhackrf version: git-f202160 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2924781b
Board ID Number: 2 (HackRF One)
Firmware Version: local-d6f9d40 (API:1.04)
Part ID Number: 0xa000cb3c 0x005d4758

leigh@boxy:~$ /usr/local/bin/hackrf_sweep -f 100:101 -1 
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 120 MHz
hackrf_start_rx_sweep() failed: Pipe error (-1000)
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .
leigh@boxy:~$ 

### Expected behaviour

The version of hackrf tools that Ubuntu 20.0.4 installs by default works fine:

leigh@boxy:~$ /usr/bin/hackrf_info 
hackrf_info version: unknown
libhackrf version: git-f202160 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2924781b
Board ID Number: 2 (HackRF One)
Firmware Version: local-d6f9d40 (API:1.04)
Part ID Number: 0xa000cb3c 0x005d4758
leigh@boxy:~$ /usr/bin/hackrf_sweep -f 100:101 -1 
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 100 MHz to 120 MHz
Stop with Ctrl-C
2021-02-11, 16:56:06, 100000000, 105000000, 1000000.00, 20, -61.11, -57.38, -67.70, -63.47, -57.90
2021-02-11, 16:56:06, 110000000, 115000000, 1000000.00, 20, -64.07, -72.09, -70.99, -75.76, -71.19
2021-02-11, 16:56:06, 105000000, 110000000, 1000000.00, 20, -59.55, -67.02, -61.73, -61.16, -61.02
2021-02-11, 16:56:06, 115000000, 120000000, 1000000.00, 20, -57.77, -63.23, -69.33, -62.02, -58.43
13.4 MiB / 1.000 sec = 13.4 MiB/second

Exiting...
Total time: 1.00019 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
leigh@boxy:~$ 

### Actual behaviour

hackrf_start_rx_sweep() failed: Pipe error (-1000)

### Version information

Linux boxy 5.8.0-36-generic #40~20.04.1-Ubuntu SMP Wed Jan 6 10:15:55 UTC 2021 x86_64 x86_64 x86_64 GNU/Linux

[154717.761050] usb 1-3: new high-speed USB device number 48 using xhci_hcd
[154717.909994] usb 1-3: New USB device found, idVendor=1d50, idProduct=6089, bcdDevice= 1.04
[154717.909999] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[154717.910002] usb 1-3: Product: HackRF One
[154717.910005] usb 1-3: Manufacturer: Great Scott Gadgets
[154717.910007] usb 1-3: SerialNumber: 0000000000000000088869dc2924781b
[155751.452452] usb 1-3: USB disconnect, device number 48
[155760.640182] usb 1-3: new high-speed USB device number 49 using xhci_hcd
[155760.789207] usb 1-3: New USB device found, idVendor=1d50, idProduct=6089, bcdDevice= 1.04
[155760.789212] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[155760.789216] usb 1-3: Product: HackRF One
[155760.789219] usb 1-3: Manufacturer: Great Scott Gadgets
[155760.789221] usb 1-3: SerialNumber: 0000000000000000088869dc2924781b

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

miek, 2021-02-11:
Which version of the firmware are you running on the device? The version string shown by `hackrf_info` there doesn't match anything I'd expect.

frinkemon, 2021-02-11:
> Which version of the firmware are you running on the device? The version string shown by `hackrf_info` there doesn't match anything I'd expect.

Yeah I just noticed this. The firmware is one included with the mayhem code. I just installed the latest release from here, and this is what happens:

leigh@boxy:~/hackrf/host/hackrf-tools/src$ /usr/local/bin/hackrf_info 
hackrf_info version: git-f202160
libhackrf version: git-f202160 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc2924781b
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005d4758
leigh@boxy:~/hackrf/host/hackrf-tools/src$ /usr/local/bin/hackrf_sweep
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
**hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)**
Usage:
	[-h] # this help
	[-d serial_number] # Serial number of desired HackRF
	[-a amp_enable] # RX RF amplifier 1=Enable, 0=Disable
	[-f freq_min:freq_max] # minimum and maximum frequencies in MHz
	[-p antenna_enable] # Antenna port power, 1=Enable, 0=Disable
	[-l gain_db] # RX LNA (IF) gain, 0-40dB, 8dB steps
	[-g gain_db] # RX VGA (baseband) gain, 0-62dB, 2dB steps
	[-n num_samples] # Number of samples per frequency, 8192-4294967296
	[-w bin_width] # FFT bin width (frequency resolution) in Hz
	[-1] # one shot mode
	[-N num_sweeps] # Number of sweeps to perform
	[-B] # binary output
	[-I] # binary inverse FFT output
	-r filename # output file

Output fields:
	date, time, hz_low, hz_high, hz_bin_width, num_samples, dB, dB, . . .
leigh@boxy:~/hackrf/host/hackrf-tools/src$

miek, 2021-02-11:
OK, that's the same problem as #828 - you'd need to build the firmware from git too, as the latest host code requires features that aren't in a release yet.

frinkemon, 2021-02-11:
> OK, that's the same problem as #828 - you'd need to build the firmware from git too, as the latest host code requires features that aren't in a release yet.

Ahh OK thanks. I'll go make the toolchain work.

frinkemon, 2021-02-11:
This has been resolved, you need to build the current hackrf_usb from here, this is easy enough to do on Ubuntu. Thanks all!

closed 2021-02-11 by frinkemon