greatscottgadgets/hackrf #830: Update to my issue in closed thread.

noviceFedora, 2021-02-12:
Reference: https://github.com/mossmann/hackrf/issues/826

### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
In my older thread mentioned at the top, I talked about the phantom noise which is shown as spike. Two users responded and explained it was dc offset. I was browsing through HackRF One videos of Sarah(from Signals Everywhere) and in few of her/his videos, I don't see that phantom noise in CubicSDR, SDR Console.

Here are those few videos: 

https://www.youtube.com/watch?v=CTzWZnNtAJQ
https://www.youtube.com/watch?v=fuwW5ZNK_0Y
https://www.youtube.com/watch?v=4rpkVhBpYPM

If it was due to DC offset, wouldn't it exist for them also?

noviceFedora, 2021-02-16:
I need an answer, I need to decide if I should get a replacement.

miek, 2021-02-18:
As explained in the other thread, the central spike is normal. Getting a replacement HackRF would not fix it.

The reason why you might not see it in those videos is that some software has an option to correct the DC offset automatically and hides that spike.

noviceFedora, 2021-02-20:
> As explained in the other thread, the central spike is normal. Getting a replacement HackRF would not fix it.
> 
> The reason why you might not see it in those videos is that some software has an option to correct the DC offset automatically and hides that spike.

They are using the same software I've used, CubicSDR, SDR Console, etc.

draeath, 2021-03-15:
I'm not familiar with them, using GNU Radio almost exclusively - but I can also say this is 100% normal for a HackRF. It's an artifact of the way an SDR device like the HackRF functions and is not an artifact of the particular hardware implementation.

If you're not seeing it on other's uses, they have enabled something to suppress it. I encourage you to ask them how they did so.

closed 2021-04-01 by miek