greatscottgadgets/hackrf #831: Compatibility with CMake < 2.8.12 will be removed from a future version of CMake.

dzbyr, 2021-02-16:
I don't understand what I should do.

I want to apply readme file under `hackrf-2018.01.1/firmware`
I installed below packages. I think those are latest
```
arm-none-eabi-binutils-2.35.1-1  
arm-none-eabi-gcc-10.2.0-1
```
However, I have problems with `cmake`

```make
cmake ..
CMake Deprecation Warning at CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/axe/Downloads/hackrf-2018.01.1/firmware/toolchain-arm-cortex-m.cmake:27 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.19.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:24 (project)

CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  /home/axe/Downloads/hackrf-2018.01.1/firmware/toolchain-arm-cortex-m.cmake:28 (CMAKE_FORCE_CXX_COMPILER)
  build/CMakeFiles/3.19.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:24 (project)

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- dfu-suffix not found: not building DFU file
-- Configuring done
-- Generating done
-- Build files have been written to: /home/axe/Downloads/hackrf-2018.01.1/firmware/hackrf_usb/build

```
I have to solve above error. I still wanted to take my chances. Of course it is low
```make
make
Scanning dependencies of target hackrf_usb_m0.elf
[  2%] Building C object CMakeFiles/hackrf_usb_m0.elf.dir/home/axe/Downloads/hackrf-2018.01.1/firmware/common/m0_sleep.c.obj
[  4%] Linking C executable hackrf_usb_m0.elf
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: cannot open linker script file libopencm3_lpc43xx_m0.ld: No such file or directory
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/hackrf_usb_m0.elf.dir/build.make:103: hackrf_usb_m0.elf] Error 1
make[1]: *** [CMakeFiles/Makefile2:323: CMakeFiles/hackrf_usb_m0.elf.dir/all] Error 2
make: *** [Makefile:103: all] Error 2

```
#### Version information
**Operating system**:OS: Manjaro Linux x86_64; Kernel: 5.10.15-1-MANJARO
**CMake version**: 3.19.4

miek, 2021-02-16:
The messages from cmake are just warnings and shouldn't cause any problems.
For the build failure, you're probably missing the libopencm3 submodule. You can update all the submodules by running:

`git submodule update --init --recursive`

dzbyr, 2021-02-18:
@miek 
Agree with u, This should but doesn't compile. 
I did install `libopencm3` package, I checked again
```paru -Ss libopencm3
community/libopencm3 r2639.90753950-1 [3.75MB 44.74MB] [Installed]
    Open Source firmware library for various ARM Cortex microcontrollers
```
I searched for a way forcing `cmake` but I coudn't found `--force option`
```terminal
cmake --help
Usage

  cmake [options] <path-to-source>
  cmake [options] <path-to-existing-build>
  cmake [options] -S <path-to-source> -B <path-to-build>

Specify a source directory to (re-)generate a build system for it in the
current working directory.  Specify an existing build directory to
re-generate its build system.

Options
  -S <path-to-source>          = Explicitly specify a source directory.
  -B <path-to-build>           = Explicitly specify a build directory.
  -C <initial-cache>           = Pre-load a script to populate the cache.
  -D <var>[:<type>]=<value>    = Create or update a cmake cache entry.
  -U <globbing_expr>           = Remove matching entries from CMake cache.
  -G <generator-name>          = Specify a build system generator.
  -T <toolset-name>            = Specify toolset name if supported by
                                 generator.
  -A <platform-name>           = Specify platform name if supported by
                                 generator.
  -Wdev                        = Enable developer warnings.
  -Wno-dev                     = Suppress developer warnings.
  -Werror=dev                  = Make developer warnings errors.
  -Wno-error=dev               = Make developer warnings not errors.
  -Wdeprecated                 = Enable deprecation warnings.
  -Wno-deprecated              = Suppress deprecation warnings.
  -Werror=deprecated           = Make deprecated macro and function warnings
                                 errors.
  -Wno-error=deprecated        = Make deprecated macro and function warnings
                                 not errors.
  --preset=<preset>            = Specify a configure preset.
  --list-presets               = List available presets.
  -E                           = CMake command mode.
  -L[A][H]                     = List non-advanced cached variables.
  --build <dir>                = Build a CMake-generated project binary tree.
  --install <dir>              = Install a CMake-generated project binary
                                 tree.
  --open <dir>                 = Open generated project in the associated
                                 application.
  -N                           = View mode only.
  -P <file>                    = Process script mode.
  --find-package               = Legacy pkg-config like mode.  Do not use.
  --graphviz=[file]            = Generate graphviz of dependencies, see
                                 CMakeGraphVizOptions.cmake for more.
  --system-information [file]  = Dump information about this system.
  --log-level=<ERROR|WARNING|NOTICE|STATUS|VERBOSE|DEBUG|TRACE>
                               = Set the verbosity of messages from CMake
                                 files.  --loglevel is also accepted for
                                 backward compatibility reasons.
  --log-context                = Prepend log messages with context, if given
  --debug-trycompile           = Do not delete the try_compile build tree.
                                 Only useful on one try_compile at a time.
  --debug-output               = Put cmake in a debug mode.
  --debug-find                 = Put cmake find in a debug mode.
  --trace                      = Put cmake in trace mode.
  --trace-expand               = Put cmake in trace mode with variable
                                 expansion.
  --trace-format=<human|json-v1>
                               = Set the output format of the trace.
[... 32 lines cut ...]
  --help-variables [<f>]       = Print cmake-variables manual and exit.

Generators

The following generators are available on this platform (* marks default):
  Green Hills MULTI            = Generates Green Hills MULTI files
                                 (experimental, work-in-progress).
* Unix Makefiles               = Generates standard UNIX makefiles.
  Ninja                        = Generates build.ninja files.
  Ninja Multi-Config           = Generates build-<Config>.ninja files.
  Watcom WMake                 = Generates Watcom WMake makefiles.
  CodeBlocks - Ninja           = Generates CodeBlocks project files.
  CodeBlocks - Unix Makefiles  = Generates CodeBlocks project files.
  CodeLite - Ninja             = Generates CodeLite project files.
  CodeLite - Unix Makefiles    = Generates CodeLite project files.
  Eclipse CDT4 - Ninja         = Generates Eclipse CDT 4.0 project files.
  Eclipse CDT4 - Unix Makefiles= Generates Eclipse CDT 4.0 project files.
  Kate - Ninja                 = Generates Kate project files.
  Kate - Unix Makefiles        = Generates Kate project files.
  Sublime Text 2 - Ninja       = Generates Sublime Text 2 project files.
  Sublime Text 2 - Unix Makefiles
                               = Generates Sublime Text 2 project files.
```

I'm not used to using **arm** packages and libraries. But **cmake** looks as a problem. I think, missing something related to it.

miek, 2021-02-18:
Installing the libopencm3 package won't help, the build requires a patched version of it to be in the `firmware/libopencm3` directory.

I just realised that you're not building from a git version, so the command I gave above won't work.
Instead you can get libopencm3 from here: https://github.com/mossmann/libopencm3/archive/6a301c81f4144d72fc881eb4fc7cc77aa1a32a67.zip and unpack it to the `firmware/libopencm3` directory.

dzbyr, 2021-02-18:
New cause some changes...
We get solved `libopencm3` problem
here modified directory
```terminal
ls -n
total 96
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 blinky
drwxr-xr-x 12 1000 1000 4096 Feb 18 23:06 build
-rw-r--r--  1 1000 1000 1228 Jan 29  2018 CMakeLists.txt
drwxr-xr-x  3 1000 1000 4096 Jan 29  2018 common
drwxr-xr-x  4 1000 1000 4096 Jan 29  2018 cpld
-rw-r--r--  1 1000 1000  241 Jan 29  2018 dfu.py
-rw-r--r--  1 1000 1000 1952 Jan 29  2018 dfu-util.cmake
-rw-r--r--  1 1000 1000 6809 Jan 29  2018 hackrf-common.cmake
drwxr-xr-x  3 1000 1000 4096 Feb 16 10:23 hackrf_usb
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 libopencm3
drwx------  7 1000 1000 4096 Oct  2  2019 libopencm3-6a301c81f4144d72fc881eb4fc7cc77aa1a32a67
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 mixertx
-rw-r--r--  1 1000 1000 1657 Jan 29  2018 README
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 sgpio
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 sgpio_passthrough
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 sgpio-rx
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 simpletx
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 spiflash
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 startup
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 startup_systick
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 startup_systick_perfo
-rw-r--r--  1 1000 1000 1967 Jan 29  2018 toolchain-arm-cortex-m.cmake
drwxr-xr-x  2 1000 1000 4096 Jan 29  2018 tools
```

 But **cmake** never satisfy.

```terminal
cmake ..
CMake Deprecation Warning at CMakeLists.txt:23 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:27 (CMAKE_FORCE_C_COMPILER)
  /usr/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:123 (include)
  CMakeLists.txt:26 (project)

CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
  CMAKE_CXX_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:28 (CMAKE_FORCE_CXX_COMPILER)
  /usr/share/cmake-3.19/Modules/CMakeDetermineSystem.cmake:123 (include)
  CMakeLists.txt:26 (project)

-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/arm-none-eabi
CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:75 (message):
  The CMAKE_FORCE_C_COMPILER macro is deprecated.  Instead just set
  CMAKE_C_COMPILER and allow CMake to identify the compiler.
Call Stack (most recent call first):
  toolchain-arm-cortex-m.cmake:27 (CMAKE_FORCE_C_COMPILER)
  build/CMakeFiles/3.19.4/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:26 (project)

CMake Deprecation Warning at /usr/share/cmake-3.19/Modules/CMakeForceCompiler.cmake:89 (message):
  The CMAKE_FORCE_CXX_COMPILER macro is deprecated.  Instead just set
[... 61 lines cut ...]

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- dfu-suffix not found: not building DFU file
CMake Deprecation Warning at startup_systick_perfo/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- dfu-suffix not found: not building DFU file
CMake Deprecation Warning at hackrf_usb/CMakeLists.txt:21 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- dfu-suffix not found: not building DFU file
-- Configuring done
-- Generating done
-- Build files have been written to: /home/axe/Downloads/hackrf-2018.01.1/firmware/build
```

miek, 2021-02-18:
You need to install the `dfu-util` package to fix the dfu-suffix errors. The deprecation warnings can be ignored.

dzbyr, 2021-02-28:
Thanks @miek for all support

where I should install cause I installed below packages one by one but error is same.
 
```terminal
community/dfu-util 0.10-1 [34.94KB 104.31KB]
    Tool intended to download and upload firmware using DFU protocol to devices connected over USB

```
or 
```terminal
aur/dfu-util-git 20210210-1 [+1 ~0.33]
    Tool intended to download and upload firmware using DFU protocol to devices connected over USB
```
Also, I cloned latest version from [gitlab](https://gitlab.com/dfu-util/dfu-util) to `hackrf-2018.01.1/firmware` directory 

Further, I realized already there is a `-rw-r--r--  1 1000 1000 1952 Jan 29  2018 dfu-util.cmake` file under  `hackrf-2018.01.1/firmware` directory.

But everything not goes well  

What I am doing wrong? I didn't get it

straithe, 2021-05-10:
@dzbyr Are you still experiencing this issue?

dzbyr, 2021-05-11:
Thanks for reminding, I tried again today, I wasn't try too hard and then looks good.

I can't try with hackrfOne right now but libs and other things looks installed  (out of city) 
```
hackrf_
hackrf_clock      hackrf_cpldjtag   hackrf_debug      hackrf_info       hackrf_operacake  hackrf_spiflash   hackrf_sweep      hackrf_transfer  
```
Is this output normal? Could you check if not connect any device?
```
hackrf_info 
hackrf_info: error while loading shared libraries: libhackrf.so.0: cannot open shared object file: No such file or directory
```

this time I also installed [libusbx](https://github.com/JoshBlake/libusbx)aside @miek `s [libopencm3](https://github.com/mossmann/hackrf/issues/831#issuecomment-781459862) recommendation
I don't know how much affected but looks good

```make
cmake ..
CMake Deprecation Warning at CMakeLists.txt:3 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
CMake Deprecation Warning at libhackrf/CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found PkgConfig: /usr/bin/pkg-config (found version "1.7.3") 
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.24
CMake Warning (dev) at /usr/share/cmake-3.20/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (LIBUSB)
  does not match the name of the calling package (USB1).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake/modules/FindUSB1.cmake:39 (FIND_PACKAGE_HANDLE_STANDARD_ARGS)
  libhackrf/CMakeLists.txt:48 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found LIBUSB: /usr/local/lib/libusb-1.0.so  
-- Looking for include file pthread.h
-- Looking for include file pthread.h - found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Setting udev rule group to - plugdev
-- HackRF udev rules will be installed to '/etc/udev/rules.d' upon running 'make install'
CMake Deprecation Warning at hackrf-tools/CMakeLists.txt:24 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Found FFTW: /usr/lib/libfftw3.so  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/.../git_clones/hackrf/host/build
```

straithe, 2021-05-27:
Great! I'll close this issue as it sounds like your issue has been solved.

closed 2021-05-27 by straithe

jinfagang, 2022-03-29:
How to remove this warning?