greatscottgadgets/hackrf #832: Will DC Offset phantom spike pose a problem in survey with rtl_power?

noviceFedora, 2021-02-16:
"The rtlsdr" dc offset varies because it's actually not really one single device due to multiple tuners and hardware versions, so comparing dc offsets is difficult, and more expensive transceiver like the ones sold by ADI obviously try to get rid of it. I suppose you could just search for "dc offset" in the rtlsdr subreddit, the reason why there is a faq entry is becausee this is the questions everyone new to sdr has been asking for the past 8 years.

That being said I don't see how you could verify anything beyond what the faq explains, considering that I'd just ensure you never see the signal you're looking for by adjusting the software you're using - but I'd suppress the dc offset as well to get around the usual dc offset question everyone asks...

_Originally posted by @Hoernchen in https://github.com/mossmann/hackrf/issues/826#issuecomment-770313768_

miek, 2021-02-18:
`rtl_power` is only for the rtlsdr dongles, so it doesn't support the HackRF.

The recommended option would be to use `hackrf_sweep`, which can output the same format as `rtl_power`. It automatically removes the DC spike from the output.

noviceFedora, 2021-02-20:
> `rtl_power` is only for the rtlsdr dongles, so it doesn't support the HackRF.
> 
> The recommended option would be to use `hackrf_sweep`, which can output the same format as `rtl_power`. It automatically removes the DC spike from the output.

Okay.

closed 2021-02-22 by miek