greatscottgadgets/hackrf #833: Hi guys, I'm trying to transmit GPS signal to a device i'm using. For that i'm useing the HackRF One device. The thing is that i'm not able to receive any GPS signal at all (verified also with GPS test app).  I attenuated the antenna output with 30db-40db ... and nothing. What am i missing ?

arikkleiman, 2021-02-16:
### Steps to reproduce
1. Using "Original" GreateScott Gadgets HackRF One
2. Generating gpssim.bin file  -> gps-sdr-sim -e brdc3540.14n -l -20,20,100
3. transmitting the gpssim.bin file  ->  hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0

### Expected behaviour
After some time i should see GPS signal in phone (Sam S10+)

### Actual behaviour
No GPS signal 

### Version information
**Operating system**:
Ubuntu server 19.04\win 10

arikkleiman, 2021-02-16:
I verified that Hackrf one actually transmits 1.57542Ghz

AndyCapp69, 2021-02-18:
I just purchased a HackRF and have the same problem.  Is this an issue with recent units?

arikkleiman, 2021-02-18:
Actually yesterday i managed to operate it.
i had to decrease the Internal crystal clock error (-C option) to -11 ...
and add 40db attenuator

Example:

hackrf_transfer -t gpssim.bin -f 1575420000 -s 2600000 -a 1 -x 0 -C -11

Another thing,

If that won't help, you'll need to install TCXO on your hackrf one board.

Let me know how it went 🙂 ... interesting

Arik

AndyCapp69, 2021-02-23:
I'm just trying the Hackrf with an external 10MHz clock (from a TCXO).  hackrf_debug indicates it is detected ([ 0] -> 0x01)
but still nothing. I think my Hackrf is transmitting, as the signals from real GPS seem to be much reduced. I've tried using my own generated files (-b 8 for hackrf) and downloaded samples.  Can't think of anything else to try.

arikkleiman, 2021-02-23:
I'm using a fixed location and my phone is right on top of the Hackrf
device.
still there are sometimes gps reception issues ... but i do see the fake
satellites

arikkleiman, 2021-02-23:
... And another thing,

From my experience, you got to have the latest brdc file

The archive for the daily file can be downloaded from:
https://cddis.nasa.gov/archive/gnss/data/daily/

straithe, 2021-02-28:
I'm going to close this ticket as it seems you have answered your own question. Great job!

closed 2021-02-28 by straithe

AndyCapp69, 2021-03-01:
I have now tried my HackRF with the correct internal TCXO.  My old Samsung now locks - sometimes.  My new Samsung at least is seeing some satellites, but wont lock (newer GPS receivers have far better filters I suppose). I was hoping to use this to test leap seconds, but that seems unlikely.

AndyCapp69, 2021-03-04:
Using circle_hackrf.bin (eg hackrf_transfer -t circle_hackrf.bin  -f 1575420000 -s 2600000  -a 1 -x 0). HackRF has a TCXO inside. 
hackrf_debug indicates it is detected ([ 0] -> 0x01)
For brief periods I get good signal values (40+) and am relocated to a McDonalds in Japan, but it's just far too unstable to be of any use.  Since I'm using the download file, I'm assuming the HackRF can't hack it. Has anybody had reliable results with a better SDR?   I've tried many different antennas