greatscottgadgets/hackrf #834: Spikes between 3470 and 3570 MHz - source for interference?

p-paule, 2021-02-19:
I'm a beginner with SDR and I'm very impressed about the versatility of the HackRF!

I had the below issue with two different HackRF and on two different systems (Lenovo E460 and Raspberry Pi 4). I also did the test in different locations (home, office, -2lvl basement) with similar results. Aside from 3,470 - 3,570 MHz there is nothing unusual on the spectrum from 1 - 6,000 MHz.

Could the spikes I see originate from the USB transfers or some clock?

### Steps to reproduce
1. Connect HackRF on USB 3 Port
2. Remove antenna from HackRF
2. Remove Charger from computer if possible (Notebook)
3. Disable all wireless interfaces on computer
4. Move HackRF at least 90cm/35inch away from computer
3. Open QSpectrumanalyzer and select soapy_power as backend with "--even --fft-window-boxcar --remove-dc" as additional parameters. Sample rate: 2.560 MHz, Bandwith: 0, LNB LO: 0, Waterfall history size: 100.
4. Set the following parameters: Start: 3,460 MHz, Stop: 3,580 MHz, Bin size: 10 kHz, Interval: 1s, Gain: 37dB, Corr.: 0, Crop.: 0. Select Main curve, Max. hold and Min. hold.
5. Right click on graph and select "View All"

### Expected behaviour
Graph close to the noise level.

### Actual behaviour
Spikes >-90db at 3,470, 3,495, 3,519, 3,542 and 3,570 MHz
[image]
[image]
[image]

### Version information
**Operating system**:
Manjaro with 5.10 kernel

**hackrf_info output:**

hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc332b8f1b
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0058435f

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

qspectrumanalyzer 2.2.0-4
soapy_power 1.6.1-8
soapyhackrf 0.3.3-2

### Output
```
Insert any commandline or build output here
```

straithe, 2021-05-13:
Sorry about not responding to this earlier. Are you still experiencing this issue?

p-paule, 2021-05-13:
Thanks for asking @straithe! Yes, I still have this issue.

p-paule, 2021-05-13:
I did run the last update on the system when I experienced the issue at the time of creating this ticket. I can run a software update next time the system is up and see if anything changes.

straithe, 2021-06-14:
Have you had a chance to run a software update and see if anything changed?

straithe, 2021-07-18:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

p-paule, 2021-07-21:
Hi @straithe, thanks for coming back on this. Just re-tested it. 
At least in my case (Manjaro) there were no updates (qspectrumanalyzer 2.2.0-4 and
soapy_power 1.6.1-8) and the issue is still there.
Is there anything else I can test?

p-paule, 2021-08-16:
Hi @straithe, unfortunately I don't have the permission to re-open this issue. Do you prefer I open a new issue instead?
Thanks for the help and have a good day.

straithe, 2021-09-27:
Can you show me the output of `hackrf_info`, please?