greatscottgadgets/hackrf #835: dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory

miek, 2021-02-22:
HI,
I have the same issue, and am trying to recover the SPI Flash Firmware in DFU mode. 

There seems to be a problem with "hackrf_one_usb.dfu"  ===>  "No such file or directory"

Please refer below:
"
andrew@pi-sdr:~/Desktop$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory
andrew@pi-sdr:~/Desktop$ 
"
Please note that I am following the directions at:
https://mirrors.spacecruft.org/mossmann/hackrf_wiki/src/branch/master/Updating-Firmware.md

_Originally posted by @4JAM4 in https://github.com/mossmann/hackrf/issues/396#issuecomment-782772567_

miek, 2021-02-22:
In that command, `hackrf_one_usb.dfu` should point to the file where it has been extracted from the [release package](https://github.com/mossmann/hackrf/releases/latest). You can either switch to the correct directory before running it with something like `cd hackrf-2018.01.1/firmware-bin` & run the above command again, or adjust the path in the command.

4JAM4, 2021-02-22:
Hi meik,

Thank you for your assistance, problem sorted.  I just needed to be in the right directory.  I have attached the txt from the terminal below.  It may help others. Note: you need to be in DFU mode, press the DFU button during power-up.
"
andrew@pi-sdr:~/hackrf-2018.01.1/firmware-bin$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c52e Logitech, Inc. MK260 Wireless Combo Receiver
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
andrew@pi-sdr:~/hackrf-2018.01.1/firmware-bin$ lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 046d:c52e Logitech, Inc. MK260 Wireless Combo Receiver
Bus 001 Device 004: ID 1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
andrew@pi-sdr:~/hackrf-2018.01.1/firmware-bin$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        22056 bytes
Download done.
dfu-util: unable to read DFU status after completion
andrew@pi-sdr:~/hackrf-2018.01.1/firmware-bin$ hackrf_spiflash -w hackrf_one_usb.bin
File size 22024 bytes.
Erasing SPI flash.
Writing 22024 bytes at 0x000000.
andrew@pi-sdr:~/hackrf-2018.01.1/firmware-bin$

miek, 2021-02-22:
Great, glad it's sorted!

closed 2021-02-22 by miek

adammanella, 2022-05-03:
is it normal to have that one line near the end....

unable to read DFU status after completion?

for some reason I can't get my H2 to show anything other than a black screen when reattaching it after flashing the HackRF standalone board.

Thank you,
A

adammanella, 2022-05-03:
also once I reset the device with the reset button, as per instructions after flashing, the USB led goes away