greatscottgadgets/hackrf #837: Unable to start HackRF One

noviceFedora, 2021-02-27:
### Steps to reproduce
1. Use the provided USB cable to connect HackRF One to a USB port on running system
2. Start GQRX or other software on Linux or Windows
3. Select and start HackRF One

### Expected behaviour
The HackRF One should start

### Actual behaviour
GQRX and other software are not detecting HackRF One, GQRX shows it is unable to get board ID (-1000)

### Version information
**Operating system**: Dragon OS 10, Windows 10 20H2 V2

**hackrf_info output:** Forgot this but I remember the firmware is 2018 one

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
GQRX 2.11.
### Output
```
Insert any commandline or build output here
```
I'm unable to start HackRF One on Windows 10 and Dragon OS, I'm connecting the HackRF One directly to the computer. When I try to open this in GQRX, I'm getting an error which shows a message like unable to get board ID (-1000), input/output error. I've tried disconnecting and reconnecting, resetting the HackRF One but none of these are resolving the problem.

straithe, 2021-02-28:
Can you send me a screenshot of your Configure I/O devices screen? (If you are in GQRX you can find this screen by clicking "File -> I/O Devices".

noviceFedora, 2021-03-01:
When I do that I'm seeing HackRF One, in Device String field I'm seeing hackrf. 

I managed to troubleshoot it by connecting it to a different computer and it ran without problems on this another computer. The old computer must be having some driver issues or USB port issues.

closed 2021-03-01 by noviceFedora

straithe, 2021-03-01:
Thank you for the update!