greatscottgadgets/hackrf #839: hackrf_sweep: feature not supported by installed firmware (-1005)

marcoboehrs, 2021-02-28:
### Steps to reproduce
1. run hackrf_sweep -f 1000:2000 on Ubuntu 20.04

### Expected behaviour
Command should run and output Data, like on my Kali installation

### Actual behaviour
I get this error message:
hackrf_start_rx_sweep() failed: feature not supported by installed firmware (-1005)
GQRX is running as expected

### Version information
**Operating system**:
Ubuntu 20.04.2 LTS

**hackrf_info output:**
hackrf_info version: git-f202160
libhackrf version: git-f202160 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000007xx0xxdcxxxxxxxx
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004c4f67

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

miek, 2021-02-28:
See #828 & #829

marcoboehrs, 2021-02-28:
So i have to build the Firmware from source? Release bin's from git are too old?

miek, 2021-02-28:
Yep (or use the release version of the host software)

marcoboehrs, 2021-02-28:
OK. Will give it a try. Thanks the quick reply!

marcoboehrs, 2021-02-28:
Hackrf_sweep is working now with firmware build from git source

closed 2021-02-28 by marcoboehrs