greatscottgadgets/hackrf #840: hackrf one continues to transmit after stopping GNU Radio flowgraph

gamecat69, 2021-03-01:
### Steps to reproduce
1. Create a simple GNU Radio flowgraph (sin wave -> multiply const (0.1) -> osmocom sink (hackrf)
2. Hit play and receive the signal using a device on another PC e.g. RTL-SDR
3. Stop the GNU Radio flowgraph - the signal continues to transmit and the hackrf TX led stays on
4. The TX can manually by stopped by running hackrf_info in a terminal

### Expected behaviour
The hackrf should stop transmitting when the GNU Radio flowgraph is stopped.

### Actual behaviour
The hackrf continued to transmit until `hackrf_info` was run in a terminal

### Version information
**Operating system**: Ubuntu 20.04

**hackrf_info output:**
```
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 000000000000000075b068dc325a1707
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005d4754
```

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.
I am using GNU Radio 3.8

GravesJake, 2021-03-11:
Have you tried transmitting manually, ie. through the command line, to see if transmission successfully stops there?

gamecat69, 2021-03-22:
Hi @GravesJake , thanks for your reply.

I can confirm that using `hackrf_transfer -t /dev/zero` does not exhibit this issue.
The device stops transmitting when the above command exits.

It therefore looks like it is a problem with either the osmocom sink, or the way the hackrf device interacts with the osmocom sink.

straithe, 2021-04-23:
Are you using the "kill" button or are you closing the foreground window when you try to stop the HackRF from transmitting?

straithe, 2021-05-23:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-05-23 by straithe

gamecat69, 2021-05-24:
Interesting…. I receive an email update when the issue is closed, but not when a question is raised that needs an answer. Something is wrong there I think!

To answer the question. When a GNU radio recipe is stopped using the GUI “stop” button , the hackrf continues to transmit.

Kind Regards,
Nik
(Sent from a mobile device)