greatscottgadgets/hackrf #841: How to hardware-disable the transmit-capability on HackRF

cjh08052, 2021-03-01:
I currently own a HackRF One and need to use it in an environment where I must hardware-disable the RF transmitter. I tried looking through post and Google-ing for any info; but I'm sure most people do not have such a restriction. Boy-scouts honour of 'I simply will not transmit' is not going to be good enough.

  Is there a way to hardware-disable the transmitter on the HackRF. I'm willing to solder/desolder and even cut traces as needed. I would prefer a reversible solution, but if one does not exist that's fine.

### Steps to reproduce
N/A

### Expected behaviour
The HackRF will no longer be able to transmit.

mossmann, 2021-03-01:
There are several ways to accomplish this, but I think the cleanest reversible method would be to desolder components C8 and L12:
[image]

Removal of C8 will block (most of) the signal transmitted by the intermediate frequency (IF) transceiver. Removal of L12 will prevent the transmit RF amplifier from being powered (which could otherwise amplify a very weak signal up to detectable levels). With these modifications, the unit will still appear to transmit, but any transmitted signal should be too weak to detect.

straithe, 2021-04-23:
How did desoldering the components work out for you, @cjh08052 ?

cjh08052, 2021-04-24:
Thanks for checking in on me and this effort. We have not gotten approval to to use the radio nor modify it yet. It's been uphill.

straithe, 2021-06-13:
I'm going to close this issue as desoldering the components should work and it seems like it might take a while for you to test things. Please re-open this issue, or open a new issue referencing this one if the provided solution doesn't work.

closed 2021-06-13 by straithe

N00b011010101010, 2026-10-09:
Update for hackrf pro, the pcd trace to disable is encircled in red, got this from their discord, thought make it public cause I can't find it anywhere else

[image]