greatscottgadgets/hackrf #842: HackRF only works after reboot

ArtisRu, 2021-03-02:
### Steps to reproduce
1. plug in HackRF
2. run hackrf_sweep -a 0 f 100:501
3. CTRL-C to stop it
4. un hackrf_sweep -a 0 f 100:501 again

### Expected behaviour
Expecting output like:
...
2021-03-02, 16:48:21, 245000000, 250000000, 1000000.00, 20, -64.51, -65.21, -61.24, -57.91, -58.98
2021-03-02, 16:48:21, 255000000, 260000000, 1000000.00, 20, -59.12, -56.67, -56.71, -56.47, -54.75
...
when running hackrf_sweep

### Actual behaviour
Only getting expected output first time running hackrf_sweep, when CTRL-C'ing to stop and running sweep again I get:
call hackrf_sample_rate_set(20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15.000 MHz)
Sweeping from 0 MHz to 6000 MHz
call hackrf_set_amp_enable(0)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00010 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

### Version information
**Operating system**: 
Ubuntu 20.04

**hackrf_info output:**
hackrf_info version: 2018.01.1
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000088869dc3937821b
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0066476c

GravesJake, 2021-03-11:
What does the exit message look like in your initial sweep? Do other hackrf functions work back to back for you?

ArtisRu, 2021-03-11:
> What does the exit message look like in your initial sweep? Do other hackrf functions work back to back for you?

2021-03-11, 19:01:38, 350000000, 355000000, 1000000.00, 20, -64.27, -65.49, -62.28, -66.98, -68.02
^CCaught signal 2
 3.4 MiB / 0.259 sec = 13.1 MiB/second

Exiting...
Total time: 2.25984 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

Other hackrf functions (such as hackrf_transfer) behave in the same way, only once, after that I have to restart my hackrf. Also no problems in windows 10 AirSpy application ...

GravesJake, 2021-03-17:
Definitely some unintended behavior there. Can you try updating your [host tools](https://github.com/mossmann/hackrf/tree/master/host) and [firmware](https://github.com/mossmann/hackrf/wiki/Updating-Firmware) to latest and repeating your test?

ArtisRu, 2021-03-22:
> Definitely some unintended behavior there. Can you try updating your [host tools](https://github.com/mossmann/hackrf/tree/master/host) and [firmware](https://github.com/mossmann/hackrf/wiki/Updating-Firmware) to latest and repeating your test?

Thanks! It works like it should now. Im guessing fixes #850, fixes #851 in hackrf_sweep.c fixed it :)

closed 2021-03-22 by ArtisRu