greatscottgadgets/hackrf #843: HaxkRF unstable

AndyCapp69, 2021-03-04:
Using circle_hackrf.bin (eg hackrf_transfer -t circle_hackrf.bin -f 1575420000 -s 2600000 -a 1 -x 0). HackRF has a TCXO inside.
hackrf_debug indicates it is detected ([ 0] -> 0x01)
For brief periods I get good signal values (40+) and am relocated to a McDonalds in Japan, but it's just far too unstable to be of any use. Since I'm using the download file, I'm assuming the HackRF can't hack it. Has anybody had reliable results with a better SDR? I've tried many different antennas

GravesJake, 2021-03-11:
Could use a bit more info. Have you tried other files to transmit? Have you played around with the gain or sample rate? What does `hackrf_info` show?

AndyCapp69, 2021-03-11:
I have generated my own files, but I thought it would complicate the issue by using them – better to use a tried and tested file until I know

it’s stable.

 

I have tried many combinations of gain – but not sample  rate.

 

I now have access to a 250 dollar TCXO/OCXO, rather than the 10 dollar one for the HackRF.   I should be able to try that next week.

 

 

hackrf_info 

hackrf_info version: git-f202160

libhackrf version: git-f202160 (0.5)

Found HackRF

Index: 0

Serial number: 000000000000000075b068dc31922607

Board ID Number: 2 (HackRF One)

Firmware Version: 2018.01.1 (API:1.02)

Part ID Number: 0xa000cb3c 0x0053435b

GravesJake, 2021-03-17:
Software/firmware looks okay, you could also try rolling back your host tools to the release version or updating your software/firmware to latest so that they're both using the same version (git vs. 1.02). Let me know how that works out for you. Are you able to use other HackRF functions with any success?

closed 2021-04-23 by straithe