greatscottgadgets/hackrf #844: Problem with band select

watchix, 2021-03-04:
Hi,

I'm having an issue with the band select ... it seems that HackRF don't select the band i suggested it.

In France FM band is 87,5MHz to 108MHz, no problem to receive/view/listen with #SDR or GQRX (Linux & Windows)
But with HackRF i can receive FM radio before this band en after ...

As example : "JAZZ Radio" : 99,2 MHZ => this frequency is normal
[https://i.ibb.co/xJFLbBP/Jazz-Radio-OK.jpg](url)

But at this frequency +20MHz i can hear the same FM station : "JAZZ Radio : 119,2MHz => Not normal
[https://i.ibb.co/Q6tNm0Y/Jazz-Radio-PLUS-NOK.jpg](url)

The same at 99,2 - 20Mhz : 79,2MHz => Not normal.
[https://i.ibb.co/1nZfzgD/Jazz-Radio-Minus-NOK.jpg](url)

With another SDR there is no problem ...

Tests : 
Differents OS and computer.
Multiple reboot and different antenna on the HackRF.

Firmware : 2018.01.1
This HackRF is a recent buy with no modifications.

Regards,

watchix, 2021-03-05:
Hi, I did a video/screen capture about the problem.

https://youtu.be/NfzjkGC7Ssw

Regards,

GravesJake, 2021-03-11:
That is strange, have you tried [updating](https://github.com/mossmann/hackrf/wiki/Updating-Firmware) your [firmware](https://github.com/mossmann/hackrf/tree/master/firmware) or [host](https://github.com/mossmann/hackrf/tree/master/host) code from source?

watchix, 2021-03-17:
Hi GraveJake,

Thanks for your reply, and so to "PassionRadio".

I've updated the SPI and CPLD with success (as i know) : No errors reported during compiling and transfer.
But now the device it K.O ...

Before udates :
[watchix-pc build]# hackrf_info 
hackrf_info version: git-f202160
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000xxxxxxxx23
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x0064475a

After updated SPI & CPLD :
watchix-pc sgpio_if]# hackrf_info
hackrf_info version: git-f202160
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000xxxxxxxx23
Board ID Number: 2 (HackRF One)
Firmware Version: git-f202160 (API:1.04)
Part ID Number: 0xa000cb3c 0x0064475a

[watchix@watchix-pc ~]$ hackrf_info 
hackrf_info version: git-f202160
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000xxxxxxxx23
Board ID Number: 2 (HackRF One)
Firmware Version: git-f202160 (API:1.04)
Part ID Number: 0xa000cb3c 0x0064475a
[watchix@watchix-pc ~]$ hackrf_transfer -t /dev/zero
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00012 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose() done
exit
[watchix@watchix-pc ~]$ hackrf_transfer -r /dev/null -s 21500000
call hackrf_set_sample_rate(21500000 Hz/21.500 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(900000000 Hz/900.000 MHz)
Stop with Ctrl-C
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 1.00020 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose() done
exit
[watchix@watchix-pc ~]$ hackrf_info 
hackrf_info version: git-f202160
libhackrf version: 2018.01.1 (0.5)
Found HackRF
Index: 0
Serial number: 00000000000000xxxxxxxx23
Board ID Number: 2 (HackRF One)
Firmware Version: git-f202160 (API:1.04)
Part ID Number: 0xa000cb3c 0x0064475a

Regards,

GravesJake, 2021-03-17:
Thanks for providing more info, it looks like before updating your firmware you were using host tools from source. Could you try rolling back both your host tools and firmware to the [latest release](https://github.com/mossmann/hackrf/releases/tag/v2018.01.1). Also, have you tried a different USB cable?

miek, 2021-03-18:
> I did a new video to expose the problem : https://youtu.be/fE7FgPdwqcc
> As you can see at the end of the video, there is a strange thing.
> The power level increase as i get closer from the 114.697MHz (sorry for my english it's late )

What you're seeing is aliasing. Because the HackRF's baseband filter rolls off gradually, strong signals just outside the visible band aren't always attenuated enough and so they "fold" back into band. That's why you see the power gradually reduce as you tune further away - the baseband filter is attenuating the signal more.

What you can do to minimise it, is adjust the receiver settings in GQRX. Rather than going with the automatic filter selection, you can set it manually with the `Bandwidth` setting. I'd recommend trying it out with the bandwidth set to half of the sample rate to see if that helps, and then you can tweak it from there for best results

watchix, 2021-03-18:
Hi miek,

Thanks for this precision, i'll test your advice 😃 
But for the moment it's not the main problem.

Regards,

watchix, 2021-03-25:
Hi,

It seems that my problem is simmilar than spurius frequencies, but only from the FM band...
I did some tests this night and i can hear some FM radios from 36MHz to 130MHz ... lol.

My antennas are near (1Km) of two FM broadcast towers (1kW to 1.5kW)
I bought a FM band reject-filter (80MHz to 110MHz) to test if the problem still continue.

See you next week to test if the problem still occur with reject-filter.

Regards,

straithe, 2021-04-23:
Did the reject-filter help or do you still need assistance?

watchix, 2021-05-17:
Hi Straithe,

With the FM reject-filter it's better but i'm still seeing and listening some FM radio sound when i'm listening in LSB/USB/NFM in low or high frequencies (2 to 40MHz and 114 to 600MHz)
I'm a little disappointed about this product.
However with same antennas there is no problems with another cheap SDR like RTL-SDR (RTL2832U)

Regards.

mossmann, 2021-05-21:
I'm certain that @miek is correct. The signal images evident in your screen captures are due to aliasing, a normal and expected phenomenon. You should be able to greatly reduce the power of those aliases by lowering the anti-aliasing filter bandwidth while maintaining a high sample rate. I recommend trying bandwidth of 2.5 MHz and sample rate of 20 Msps.

While rtl-sdr devices were designed with filtering and hardware configuration specifically tuned for FM broadcast usage, HackRF One is designed for more general use. There are configuration trade-offs in your control as a HackRF user that are not in your control as an rtl-sdr user. To learn more about aliasing and other things that can help you use your HackRF more effectively, I recommend watching my [educational videos on SDR](https://greatscottgadgets.com/sdr/).

closed 2021-06-14 by straithe