greatscottgadgets/hackrf #845: HackOne RF Bricked?

visseroth, 2021-03-11:
I bought a HackOne RF a while back to tinker with and I was going to use it recently to do some RF sniffing with a friend to find some local interference but when I plugged it in  I noticed my computer wasn't detecting the device.

The power light comes on but the other lights are quite dim.
3v3, bright
1VB, very dim
RF, No light
USB, very dim
RX, Dim
TX, Dim

No matter what machine I plug it into, Linux, Windows, different machine, the results are the same.

I've also tried different USB cables, results are the same.

Journalctl -f on my linux machine does not indicate any new device is plugged in when plugging in the HackOne.

Pressing of the buttons have no effect.

Did I miss anything else?

Suggestions?

miek, 2021-03-12:
Does it come up in journalctl if you start it in DFU mode? (by holding the DFU button when plugging it in)

visseroth, 2021-03-12:
I just tried it and no, there was no response on the lights or journalctl

visseroth, 2021-03-22:
Anything?

miek, 2021-03-31:
I think this was answered by email in the end? But, if not, this does sound like a hardware failure and you can get a replacement by getting in touch with the distributor you bought from.

visseroth, 2021-03-31:
Thanks for the reply.
I contacted the seller and they worked with me on getting it RMA'd.
Unfortunately nothing I tried seem to get any response from the unit so I'll be sending it in.

closed 2021-03-31 by miek