greatscottgadgets/hackrf #846: Burned out LXES1TBCC2 chip

akadooie, 2021-03-12:
I bought a HackOne RF a while back to tinker with and I was going to use it recently to do some RF sniffing but I noticed that the device would not connect trough USB anymore. After some investigation I found out that the ‘LXES1TBCC2-004’ was burned out. 

The power light comes on.
3v3, bright
1V8, bright
RF, bright
USB, No light
RX, No light
TX, No light

Now I am wondering how this could happen? I could be I did something wrong, but I don’t know what.
Should I be able to fix this issue if I can find a replacement chip? If so, are there any chips you guys suggest? The one which was on there seem quite hard to find/order (I’m living in the Netherlands).

Ps. if a picture would be helpful let me know and I will post some.

Thanks in advance!

miek, 2021-03-12:
Yeah, some photos would be helpful - thanks.

The only thing I can think would cause that IC to fail is over-voltage, possibly from a misbehaving USB-PD device.
It's possible that other parts downstream might've got damaged too. You can try just removing that part, and testing again - the HackRF will work without it but won't have ESD protection.

That part was switched to VBUS54CV-HSF-G4-08 a few years ago - that should be easier to find. :)

akadooie, 2021-03-13:
The device works (or seem to work) when I use a portapack but that's a bit limited. Besides that I have no way to upgrade the firmware :-(

Most of the time I use an "Aukey 30000mAh Power Bank".

Here is a picture of the where the burned out chip was (I have cleaned it up a bit), the chip itself just kinda went to dust when i poked it with a small piece of plastic.
[image]

N8AUM, 2021-03-14:
Wounder how that resister blew ? Isn't that going to one of the com pins on USB port ?

straithe, 2021-04-24:
The two traces that go around the part that has been removed are the data traces. If those two traces are damaged it might account for the limited ability that you are seeing. To test this, use a continuity tester to check that there is continuity from the USB connector to the pair of 0 ohm resistors. Let me know what you find!

akadooie, 2021-05-20:
Sorry for taking my sweet time to get back to you.

I have tested a few traces and they seem to be okay (at least there is continuity). Attached a picture where I marked the traces that I have tested in yellow. All those traces have continuity, wouldn't that be strange?
[image]

cjheath, 2021-05-21:
No, that's not strange. The two chip inductors are ferrite beads in series with the power lines, which should measure as a dead short on a DC meter. The centre two traces are also meant to pass through to the CPU chip, with the protection device going to the rails.

I'm afraid that if that protection diode got smoked (which took a fairly big hit!), then so did your CPU chip's USB peripheral, and you'll need to replace that chip to revive your HackRF. The big surprise to me is that the CPU isn't completely dead.

straithe, 2021-05-21:
You could try replacing the USB connector or trying different USB cables, but it is more likely that @cjheath is correct in their device diagnosis and you might want to try replacing the CPU.

akadooie, 2021-05-22:
@cjheath Thanks for the clear explanation!

I don't know if my soldering skills are up to the task of replacing the CPU... seems like quite a bit of delicate work lol
Then there is also the risk that more components have taken a hit and died?

cjheath, 2021-05-23:
Sorry, I thought I had read that the HackRF was still working (e.g. with a Portapack) but not on USB. This would indicate that the device was killed by mischief on the data lines. If it's not working at all, then yes, there is a fair chance of other damage, because it's much more likely that the damage came from the power rails (it took quite a lot of heat to vaporise that protection chip). Sorry for the bad news!

akadooie, 2021-05-23:
@cjheath Yes, the device works with a Portapack! I can receive & transmit with the device while using the Portapack (powered by usb..).

But since I have no USB connection, I cannot update the firmware or simply use GNU Radio. That’s why I asked the question here, to see if there is a way to restore the USB connectivity. 
But from what I understood, I need to replace the whole CPU to fix the USB connection. Or am I wrong?

cjheath, 2021-05-24:
Well if you're sure that the USB wiring (including the socket) is intact and working, then I reckon it's worth changing the CPU. A new chip is a lot cheaper than a whole new HackRF. But that's me, I'd change the chip myself. You might find a member of a local radio club who is experienced in SMD rework and will do it for you, otherwise a phone repair shop probably can.

straithe, 2021-06-14:
I'm closing this issue as replacing the CPU should solve this issue. Please re-open this issue or open a new one if you need, @akadooie.

closed 2021-06-14 by straithe