greatscottgadgets/hackrf #849: USB PID VID are '0000' Device not recognised.

Commander-Crash, 2021-03-18:
### Steps to reproduce
1. plug in hackrf to computer/phone
2. USB PID VID are '0000' windows acts like it sees it but unknown device.
3. linux does not see device.

### Expected behaviour
I expect it to work in windows when pluged in like it has in the past.
USB light to light up

### Actual behaviour
Windows makes the device sound after 5 seconds but unknown device
No USB light on hackrf.
USB PID VID are '0000' in windows.
### Version information
Windows7-10/linux/Android8.1

**hackrf_info output:**
No hackrf -5 error in linux

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```
I have a portapack h2 running Mayhem 1.3.0  works fine. When I wanted to use the spectrum analyzer in windows is when i notice that the hackrf one stopped responding to usb. It charges the portapack. I used all my usb cables all 47 of them 10 of them brand new. I tried this with 3 other PC's with same results. This is the 2nd hackrf that has had issues in less then 6 months. Unfortunately I can not return this one due to ebay policy. No Damage is on the board that i can see. I am not able to upgrade the firmware now. I do not own a Logic Analyzer to check D- or D+

Commander-Crash, 2021-03-19:
Never mind. Im not going to get any answers hear.

Commander-Crash, 2021-03-19:
You can closed this post I don't care to continue purchasing failed hardware.

straithe, 2021-03-19:
I'm sorry to hear about your technical issues. Have you tried using Portapack rather than Portapack H2?

straithe, 2021-03-19:
Closing the issue at the user deleted themself.

closed 2021-03-19 by straithe