greatscottgadgets/hackrf #854: Help with jtag firmware upload (hackrf + portapack)

sfrmattos, 2021-03-19:
I have been trying this for weeks before asking for help, as this is not an issue.

In short, I have two hackrfs, one working normally and the other with a dead USB port.

On the device with no USB connection, I would like to try to load a firmware from the portapack. I don't know much about electronics, but I have a device capable of accessing via JTAG (black-magic-probe), and I've already managed to access the MCU via JTAG, but I'm completely lost on how to upload firmware in hackrf. As I understand it, it would have to be through ELF files, loading via "GDB load". I even managed (in theory) to load the firmware into the MCU, but apparently when I turn off the hackrf it reloads the firmware that was in it before the USB port died (I may be wrong about that).

Any help will be extremely valuable. I am doing this not only to be able to continue using the device with the dead USB port, but also for learning purposes, and I believe this information will also be valuable to many others in the same situation as me.

Thanks!

straithe, 2021-03-20:
You did an awesome job of getting this far! Thanks for sharing your journey.

At this time we do not have a direct solution for you. There are a few things you can try:
1) You can load the code into RAM every time you want to use the device, as you have been doing. 
2) You might be able to directly load the code into flash memory using something like a GreatFET or Bus Pirate.
3) Theoretically, you can write a flash memory updater that runs on the MCU and is loaded by JTAG.
4) You can repair your USB connection. 

Hopefully one of these ideas can work for you. Let me know how it works out!

sfrmattos, 2021-03-20:
Thank you @straithe. I'll keep trying to find a way to upload the firmware. Hope to get back and post a tutorial for the people on the same situation.

straithe, 2021-03-20:
Thanks! Ping me if you write a blog post on the topic.

straithe, 2021-03-24:
I'm going to close this issue as I've offered all the support I can. Please do leave updates on this ticket if you would like to share!

closed 2021-03-24 by straithe