greatscottgadgets/hackrf #858: release no longer includes cpld in firmware-bin

ZeroChaos-, 2021-03-22:
yes, I know that the hackrf code no longer expects the cpld to be flashed.  However, at this time, other code such as the portapack firmware does expect the cpld to be flashed.  

Is firmware/cpld/sgpio_if/default.xsvf the required file, or do I need to do something with it first?

mossmann, 2021-03-22:
Yes, firmware/cpld/sgpio_if/default.xsvf is the correct file.

ZeroChaos-, 2021-03-22:
thanks.  I can (and will) use this release as is, but you might want to copy the file to the firmware-bin for releases as was done previously.  There are likely to be use cases for having it, even though you have eliminated the need in the main firmware.  super excited about the release, thanks again for that!

closed 2021-03-22 by ZeroChaos-

mossmann, 2021-03-23:
After confirming with @jboone that PortaPack also loads CPLD bitstreams into SRAM from the ARM firmware, I think it is safe to continue leaving the bitstream out of release packages.