greatscottgadgets/hackrf #859: Error -1000 when updating firmware

shinobirain123, 2021-03-22:
Hi Everyone, 

  I am trying to update the firmware for my hackrf. The reference website I am using is here: [https://www.securesolutions.no/updating-the-hackrf-ones-firmware/](url).

Following steps 4 and 5, I received the error: 
"hackrf_init() failed: Other error (-1000)"

I have looked into -1000 and believe it is a libhackrf not being installed so I ran 

"sudo apt-get install libhackrf-dev"
which installed libhackrf-dev; however, using the commands shown in step 4 and 5 in the link above still result in the same error stated. The same result also occurs when using:
"hackrf_info."

GravesJake, 2021-03-22:
I recommend looking at [Operating System Tips](https://github.com/mossmann/hackrf/wiki/Operating-System-Tips) and [Software Setup](https://github.com/mossmann/hackrf/wiki/Getting-Started-with-HackRF-and-GNU-Radio#software-setup) to ensure all prerequisites are met, then following [Updating Firmware](https://github.com/mossmann/hackrf/wiki/Updating-Firmware).

What OS are you using?

shinobirain123, 2021-03-23:
Hi GravesJake!

   Thank you for taking the time to answer. I was running windows 1808 using ubuntu off of the windows store (so a vm?). 
   
   I believe I was able to go through the steps on the link I provided by using a native linux machine that I have so I do not have anymore issues here.

straithe, 2021-03-24:
It sounds like your HackRF issues have been resolved, so I am going to close this issue. If you need any more help please reopen this issue or open a new one.

closed 2021-03-24 by straithe