greatscottgadgets/hackrf #860: GNU Radio 3..9.0 Osmosdr Not Working With HackRF One Also osmocom dot org Lists As Obsolate

Richard628, 2021-03-23:
### Steps to reproduce
1.Install OS Ubuntu 20.04
2.install hackrf one library and utilities then prerequisites for gnuradio and osmosdr 
3.apt install or use cmake to install gnuradio
4. install prerequisites for osmosdr
5. install osmosdr with cmake
6. lanuch gnu radio from GUI
7. select osomsdr source sink and FFT frequency display widgets and connect. 
8.  save and run result error code -7
9. check osmocomdot org list obsolate driver
10. tried GNU RAdio UHD Source result cant find serial=0

### Expected behaviour
Should display frequency 

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

GNU Radio Ver 3.9.0
### Output
Gnu Radio Osmosdr Source
>> Done (return code 1)

Generating: '/home/richard/test.py'

Generating: '/home/richard/test.py'

Executing: /usr/bin/python3 -u /home/richard/test.py

gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
built-in source types: file rtl_tcp rfspace redpitaya 

>>> Done (return code -7)

######################
Gnu Radio UHD SOURCE
>> Done (return code 1)

Generating: '/home/richard/test.py'

Generating: '/home/richard/test.py'

Executing: /usr/bin/python3 -u /home/richard/test.py

[INFO] [UHD] linux; GNU C++ version 9.2.1 20200304; Boost_107100; UHD_3.15.0.0-2build5
Traceback (most recent call last):
  File "/home/richard/test.py", line 196, in <module>
    main()
  File "/home/richard/test.py", line 174, in main
    tb = top_block_cls()
  File "/home/richard/test.py", line 83, in __init__
    self.uhd_usrp_source_0 = uhd.usrp_source(
RuntimeError: LookupError: KeyError: No devices found for ----->
Device Address:
    serial: 0

>>> Done (return code 1)

```
Insert any commandline or build output here
```

miek, 2021-03-29:
> Executing: /usr/bin/python3 -u /home/richard/test.py
> 
> gr-osmosdr 0.2.0.0 (0.2.0) gnuradio 3.8.1.0
> built-in source types: file rtl_tcp rfspace redpitaya

This line should include `hackrf` if gr-osmosdr was built with HackRF support. I suggest looking at the cmake output when building gr-osmosdr, to see why it isn't detecting libhackrf.

Richard628, 2021-03-29:
Mike

Thank you for the for this information.  I have removed and reinstalled gr-osmosdr a couple of times even with purge option.  Two things I have noticed during installation.

After installation the gnuradio shows version 3.9 or above even after trying to roll back to version 3.8

The osmocom wiki website shows that gr-osmosdr works with gnuradio version less than 3.9
https://osmocom.org/projects/gr-osmosdr/wiki
Build process
*The Gnu Radio block requires a recent stable gnuradio (>= v3.8 and <v3.9) if building master branch.

Questions

  1.  Do you know if I can run gnuradio 3.9 or above while using the HarckRF One with gr-osmosdr?

  1.  Should I be installing gr-osmosdr using the master branch or some other new version that might be alpha or beta code.

Comment
Also by the way I am half way through your software defined radio online video course.   Wanted to say it is the best on the web.  The information was presented in a way that I understood exactly how it worked.  Thank you for taking the time to put together such a valuable resource.  It was the main reason I purchased the HackRF One.

Richard

** WARNING EXTERNAL EMAIL **

CONFIDENTIALITY NOTICE: This e-mail may contain information that is privileged, confidential or otherwise protected from disclosure. If you are not the intended recipient of this e-mail, please notify the sender immediately by return e-mail, purge it and do not disseminate or copy it.

miek, 2021-04-01:
> 1. Do you know if I can run gnuradio 3.9 or above while using the HarckRF One with gr-osmosdr?
> 1. Should I be installing gr-osmosdr using the master branch or some other new version that might be alpha or beta code. 

I don't know - it seems like there is no official support for 3.9 in gr-osmosdr yet, but it does look like there has been some work on the master branch for GNURadio 3.9 compatibility so that branch is the best to try.
I don't think the errors you have hit so far are related to 3.9 though.

Richard628, 2021-04-01:
Mike

Thanks…  I should have sometime this weekend to try the steps in your first email…

Richard

  2.  Should I be installing gr-osmosdr using the master branch or some other new version that might be alpha or beta code.

straithe, 2021-06-14:
@Richard628 , have you had a chance to try the steps suggested by miek?

closed 2021-07-18 by straithe