greatscottgadgets/hackrf #865: hackrf_sweep: sweep count issues

gozu42, 2021-03-24:
seeing this ... https://github.com/mossmann/hackrf/pull/856 
... reminds me sweep count is broken in at least one more way: 
when cranking up num_samples, each block for frequencies[0] is increasing sweep count by one. 
(== a single sweep with three data blocks per tuning is counted as three sweeps)

(and that probably causes problems with ifftw mode and anything else in this "is this a start of a sweep" code block too)

trivial fix would be to track the last_freq and only jump in there if current is freq[0] and last was not.

this would break sweep for a usecase that does not actualy retune, but i dont think that is even possible with interleaved mode currently and slightly silly to use hackrf_sweep in this case to begin with.

mossmann, 2021-03-30:
It is a bit unusual to use `hackrf_sweep` over such narrow bandwidth that it doesn't retune, but I wouldn't call it silly. It is a convenient way to get frequency domain data that isn't otherwise available from command-line tools we provide. I've definitely done things like `hackrf_sweep -f 2420:2430 -w 100000 -1` to get a quick snapshot of a small frequency range.

It wouldn't be a huge loss to remove support for small frequency ranges, but I'd rather keep that feature if we can.

gozu42, 2021-03-30:
interesting point about the lack of other cli tools. (i tend to use hackrf_sweep only for ... sweeping. and soapy_power for more focused views)

the usecase for a quick view with -1 would still work, open-ended sweep-on-just-one-tune too (but would report bogus sweep stats), what wouldnt work is something like `hackrf_sweep -f 2420:2430 -w 100000 -N 1000` (well, it would work, but not stop).
dont see an elegant way around this right now, bringing back the hostside "ranges to tuning steps" conversion logic just to detect it as a special case is certainly possible, but rather ... unattractive.
will think about it some more.

an example/demonstration of the current problem:
```
$ hackrf_sweep -1 2> /dev/null | wc -l
1200
$ hackrf_sweep -1 -n 16384 2> /dev/null | wc -l
2
$ hackrf_sweep -1 -n 32768 2> /dev/null | wc -l
2
```
these are numbers of CSV generated, the first result is correct, the other two ... less so.

gozu42, 2021-03-30:
now that i have actually run the "narrow quick view" example, i am back to "i dont think that is even possible with interleaved mode" from the initial post, because interleave mode means even a `hackrf_sweep -f 2420:2421 -1` _does_ retune:
```
$ hackrf_sweep -f 2420:2421 -1 2> /dev/null
2021-03-30, 20:30:54.177325, 2420000000, 2425000000, 1000000.00, 20, -68.40, -68.65, -60.54, -62.95, -69.49
2021-03-30, 20:30:54.177325, 2430000000, 2435000000, 1000000.00, 20, -68.26, -62.79, -60.63, -68.60, -68.37
2021-03-30, 20:30:54.177325, 2425000000, 2430000000, 1000000.00, 20, -63.24, -64.40, -65.04, -61.96, -67.14
2021-03-30, 20:30:54.177325, 2435000000, 2440000000, 1000000.00, 20, -61.11, -67.44, -71.70, -66.80, -62.00
```

so that usecase would still be fully supported, and i would leave a comment about counting sweeps in the code for whoever may add special narrow-range support to sweep (firmware side) or bring back linear sweep mode (host side) in the future.

mossmann, 2021-11-13:
Even if we were to fix the problem with incrementing `sweep_count` too often, I'm not sure that allowing `num_samples` to be larger than a single block is useful. We don't concatenate multiple blocks into a single FFT; we would run a separate FFT for each block and produce output each time. (The firmware currently replaces a few samples at the start of each block with a frequency header, so some tricky changes would be required to allow the samples in multiple consecutive blocks to be contiguous.)

The only potential advantage of having high `num_samples` is that there would be more time spent capturing samples (and more rows of output) at each tuning step with less overall time spent tuning. The cost of this would be a reduction of sweep rate. I don't see this as being particularly useful unless you were doing something like averaging together several FFTs (as proposed in #862).

#862 could be implemented without dwelling for multiple blocks provided that you don't mind restricting the maximum number of FFT bins.

I propose that we eliminate the `-n` option and use `SAMPLES_PER_BLOCK` in place of `num_samples`.

PR #993 (2021-11-15): hackrf_sweep: simplify dwell time, bin width, and time stamps
This removes support for longer dwell times at each frequency, something that I don't think has ever worked except perhaps in pre-release versions of hackrf_sweep.

fixes #865
fixes #951
closed 2021-11-15 by miek