greatscottgadgets/hackrf #866: TXCO troubleshooting tips?

5ch4um1, 2021-03-24:
### Steps to reproduce
1. plug in TXCO, connect hackrf to PC
2. hackrf_debug --si5351c -n 0 -r

### Expected behaviour
Tell us what you expect should happen
[  0] -> 0x01
### Actual behaviour
Tell us what happens instead
[  0] -> 0x51
### Version information
**Operating system**:
Lubuntu 18.04
**hackrf_info output:**
hackrf_info version: 2021.03.1
libhackrf version: 2021.03.1 (0.6)
Found HackRF
Index: 0
Serial number: [some serial number]
Board ID Number: 2 (HackRF One)
Firmware Version: 2021.03.1 (API:1.04)
Part ID Number: 0xa000cb3c 0x0041475c

I was wondering what could one do if the txco does not get recognized?
Unfortunately i have only one hackrf and one txco, so i can't easily check if it is a hardware issue.
Or, first of all, should it get recognized automatically? Could i try writing 0x01 to [ 0] with hackrf_debug?
Anything i could check with a cheap multimeter or a fx2 logic analyzer? Is it normal that those R1, R2 and C3 are not present?
Any help appreciated. 
[image]

miek, 2021-03-24:
It does detect automatically if the signal is there, so trying to force it wouldn't help.
For debugging, the best would be to use a scope on pin 2 but using the fx2 instead on its highest sample rate should work too. You should see a 10MHz square wave.

However, I think I see the problem from the photo. It looks like the 0 ohm resistor on the clock output has been installed in the wrong spot, so it doesn't bridge the gap in the trace:
[image]

5ch4um1, 2021-03-24:
Amazing help, thank you very much:
 hackrf_debug --si5351c -n 0 -r
[  0] -> 0x01

I won't post a picture of my soldering, tried to place just a little blob and desoldered the whole thing, but managed to put it back in place, and it works now... :)

closed 2021-03-24 by 5ch4um1