greatscottgadgets/hackrf #867: Baseband frequency and number of sample HackRF?

efakhri, 2021-03-31:
Hi, I'm trying to implement radar interferometry using two HackRF with center frequency 915 MHz and sampling frequency 10 MHz. I need to know the baseband frequency and the number of samples. I found the algorithm for tuning the mixer and transceiver like this:
```
if(freq_mhz < MAX_LP_FREQ_MHZ)
	{
		rf_path_set_filter(&rf_path, RF_PATH_FILTER_LOW_PASS);
#ifdef RAD1O
		max2837_freq_nominal_hz = 2300000000;
#else
		/* IF is graduated from 2650 MHz to 2343 MHz */
		max2837_freq_nominal_hz = 2650000000 - (freq / 7);
#endif
		mixer_freq_mhz = (max2837_freq_nominal_hz / FREQ_ONE_MHZ) + freq_mhz;
		/* Set Freq and read real freq */
		real_mixer_freq_hz = mixer_set_frequency(&mixer, mixer_freq_mhz);
		max2837_set_frequency(&max2837, real_mixer_freq_hz - freq);
		sgpio_cpld_stream_rx_set_q_invert(&sgpio_config, 1);
```
and I tried to calculate the baseband frequency using this algorithm. but, i'm stuck at function max2837_set_frequency and can't figure out the tune_freq.

martinling, 2021-11-15:
Hi, sorry it's taken a long time for this to be picked up.

Although you said you were looking for the baseband frequency, it sounds like what you're actually trying to find is the intermediate frequency (IF) - that's what that code you've quoted is calculating. The baseband frequency on HackRF is zero - the signal from the IF is mixed down to 0Hz so that it can be sampled by the ADC in the 0-20MHz range.

The algorithm that's used to choose the IF is a bit complicated, as you've noticed. I'm afraid there isn't an easy way to determine what IF the firmware has chosen other than following through the code, although there is some work in progress on retrieving this, which is being tracked in issues #906 and #129.

In the meantime, if you need to know the exact IF in use, the best way to do this is to use explicit tuning. If you're collecting data using hackrf_transfer, this is done using the `-i`, `-o` and `-m` options instead of just setting the centre frequency with `-f`. Use hackrf_transfer -h to see all the options available and the accepted ranges.

martinling, 2021-12-03:
I'm going to close this issue now but feel free to reopen or start a new issue if you have further questions.

closed 2021-12-03 by martinling